projecthorus/radiosonde_auto_rx #4: Receive more than one sonde signal at once

TheSkorm, 2017-12-16:
As a user I would like to receive multiple sondes at once when BoM have a failed launch and need to use another frequency

We should be able to multiplex rtl_fm to receive multiple sondes at once. It might be hard to do detection though.

TheSkorm, 2018-01-07:
Rough thoughts on this one.

#18 Needs to be done first to allow multi devices.
#36 Will probably need doing to allow different configs for each receiver.

Once those two are done if we add [pid](https://github.com/trbs/pid) library to the project. After auto_rx performs a scan but before it tries detect on each frequency it tries to open a lock file for that frequency. If it can open a lock it starts decode, if it can't it'll pass. pid library should also allow us to detect and clear broken processes. That way we can run n number of auto_rx's.

darksidelemm, 2018-05-26:
Working on this under https://github.com/darksidelemm/radiosonde_auto_rx/blob/restructure/auto_rx/

darksidelemm, 2018-06-02:
Supported in https://github.com/projecthorus/radiosonde_auto_rx/commit/f8877111c805cf826519152f8b533b237f7a0c52

closed 2018-06-02 by darksidelemm