projecthorus/radiosonde_auto_rx #18: Allow specifying a RTLSDR Device Number

darksidelemm, 2017-12-17:
Allow specifyng a RTLSDR device number (-d N) when calling rtl_power / rtl_fm, to allow the use of multiple RTLSDRs on one system.

darksidelemm, 2017-12-17:
This might be a bit harder actually... how do we find the right dongle to reset when we run reset_usb? At the moment we just reset all of them...

Keskebeu, 2018-01-11:
lsusb will tell you which usb is connected and what actually running.

Keskebeu, 2018-01-11:
https://github.com/soldair/reset-usb/blob/master/README.md

You can use this tool to restart a particular one

darksidelemm, 2018-01-11:
Understood - already do this with a single dongle.

The issue is tying the RTLSDR 'device index' to the USB bus and device index. 

It's possible that the more detailed output that can be provided by lsusb will tell us this information.

Keskebeu, 2018-01-12:
Is your question running this softwate in multiple dongles?

darksidelemm, 2018-01-12:
The eventual aim (and one of the reasons this issue was created) was to eventually allow use of multiple SDR dongles, which which would allow for a few different use-cases. For example, with two dongles, one could be continually scanning, and the other tasked onto a detected radiosonde.

Part of the move toward this kind of functionality is to enable the user to define which SDR dongle to use, hence this issue. 
As we know there are common issues with RTLSDR devices glitching and requiring a reset, we need to make sure we can determine the usb bus/device number of the RTLSDR currently in use, and only reset that single device - not another RTLSDR which may be in use tracking another radiosonde.

Keskebeu, 2018-01-12:
You can run lsusb -v or lsusb -D and will give you more info about the device. 

usb-devices is also another option. You can always do : cat /sys/bus/usb/devices/*/product

Or go directly to usb directory.

Keskebeu, 2018-01-12:
https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line

This is also really good.

Keskebeu, 2018-01-12:
@darksidelamm (sudo) ./reset_usb /dev/bus/usb/001/006
*/ 

This is found at the scan file. It tells you to execute lsusb and than the command to reset a particular dongle depending what you using you need to change the info.

ED6E0F17, 2018-01-28:
> It's possible that the more detailed output that can be provided by lsusb will tell us this information.

lsusb can read the serial number on a dongle, which can be matched to the output of rtl_power, or used as an input to rtl_power.
```
pi@localhost:~$ lsusb -d 0bda:2838 -v |grep Serial -B2
  iManufacturer           1 Realtek
  iProduct                2 RTL2838UHIDIR
  iSerial                 3 2
pi@localhost:~$ rtl_test -p -d 2
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 2

Using device 0: Generic RTL2832U OEM
```
Some people might want to have multiple dongles on different antenna running adsb, aprs etc, so being able to choose a device by serial number would be an advantage.

darksidelemm, 2018-05-26:
As an update, use of multiple RTLSDR dongles will be supported in a major update to the auto_rx codebase which should be ready in a few week. 
The update is essentially a re-write, and makes handling multiple sdrs much much easier. You will be able to specify a list of RTLSDR device IDs to use (each with individual gain/ppm/bias settings), and they will be tasked either either scanning, or decoding a sonde signal. 
I'm working on this update here: https://github.com/darksidelemm/radiosonde_auto_rx/blob/restructure/auto_rx/

darksidelemm, 2018-06-02:
Supported in https://github.com/projecthorus/radiosonde_auto_rx/commit/f8877111c805cf826519152f8b533b237f7a0c52

closed 2018-06-02 by darksidelemm