projecthorus/radiosonde_auto_rx #27: Payload Position Sanity Check

darksidelemm, 2017-12-22:
We often see ridiculous jumps in position on RS92 sondes, due to bad calculated positions. Short of fixing the position calculation code (urghhhh), a simpler solution is to have some kind of filter on the position data.
Options include:
- Discard if change in position is too great (dangerous if we get a bad position as the first packet)
- Discard if greater than a certain radius from the receiver location (probably better)

Both of these options can result in limiting the capabilities of this receiver system, but it's probably better than spamming the tracker with bad positions. A radius of 1000km would reasonably cover all radiosonde flights, and limit most of the bad positions.

darksidelemm, 2017-12-22:
Basic position filtering added in https://github.com/projecthorus/radiosonde_auto_rx/commit/4bbec6323ba0fa46d323884a9e8d03d3fb23f2f3

darksidelemm, 2017-12-28:
Position filtering seems to do the job!

closed 2017-12-28 by darksidelemm

pit711, 2019-01-25:
i still have sometimes huge jumps with DFM sondes

https://imgur.com/a/g88xFju

darksidelemm, 2019-01-25:
Yep, we've noticed!
Until we solve the issue properly (likely just to do with a lack of CRC checking in the DFM decoder), perhaps look at changing the max_radius_km config file parameter to something smaller.

pit711, 2019-01-25:
https://i.imgur.com/PHka0WN.png
i set it to 800, but got this error

https://i.imgur.com/cMaplop.png

darksidelemm, 2019-01-25:
Could you please email me your entire config file, to vk5qi@rfhead.net
This will help debug the issue.

pit711, 2019-01-25:
> Could you please email me your entire config file, to vk5qi@rfhead.net This will help debug the issue.
> […](#)

i already deleted my reply, was my mistake, deleted a # :)

darksidelemm, 2019-01-26:
Are you Ok with me closing this issue?