projecthorus/radiosonde_auto_rx #34: Multiple data strings from the same Sonde should be in one file

zanco, 2018-01-04:
Last night I had the first auto_rx results from a sonde 60 - 100 km away from me.

I see 4 items in the log with the same sonde serial number each containing one single string of data.

Guess it would be better if those go in one file together ?

Thanks,
Ben

`-rw-r--r-- 1 debian debian     82 Jan  4 00:03 20180104-000305_M4530114_RS41_403900.log
-rw-r--r-- 1 debian debian     83 Jan  4 00:29 20180104-002941_M4530114_RS41_403900.log
-rw-r--r-- 1 debian debian     83 Jan  4 00:35 20180104-003549_M4530114_RS41_403900.log
-rw-r--r-- 1 debian debian     83 Jan  4 00:39 20180104-003918_M4530114_RS41_403900.log
`

darksidelemm, 2018-01-04:
I did consider this when writing the per-sonde log support. A new log file is started when the sonde decoder function gets its first valid telemetry frame. Hence, if you have a consistent 'lock' on a sonde you will only end up with one log file.

I think we can probably add in something to look in the log directory for a filename containing the current serial number...

darksidelemm, 2018-01-04:
Fixed in https://github.com/projecthorus/radiosonde_auto_rx/commit/b46758b0dce0359896a8f8d1204d28dff1f98d3e

closed 2018-01-04 by darksidelemm