projecthorus/radiosonde_auto_rx #38: Support for AirSpy Mini ( airspy_rx instead of rtl_fm )

vk4tec, 2018-02-11:
Hi Mark,

I have an AirSpy Mini I would like to utilise for this instead of an RTL-SDR

Airpspy mini uses a tool called "airspy_rx" instead of "rtl_fm"

I think you are just using rtl_fm and rtl_power ?

The options are shown below for airspy_rx

Do you think you could do me hack, My radio sondes are on 401.500 MHz 

airspy_info reveals 

airspy_lib_version: 1.0.9

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc9-0-ga56adfd 2016-06-12
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x644064DC324645CD
Supported sample rates:
        6.000000 MSPS
        3.000000 MSPS
Close board 1

//

airspy_rx v1.0.5 23 April 2016
Usage:
-r <filename>: Receive data into file
-w Receive data into file with WAV header and automatic name
 This is for SDR# compatibility and may not work with other software
[-s serial_number_64bits]: Open device with specified 64bits serial number
[-p packing]: Set packing for samples,
 1=enabled(12bits packed), 0=disabled(default 16bits not packed)
[-f frequency_MHz]: Set frequency in MHz between [24, 1900] (default 900MHz)
[-a sample_rate]: Set sample rate
[-t sample_type]: Set sample type,
 0=FLOAT32_IQ, 1=FLOAT32_REAL, 2=INT16_IQ(default), 3=INT16_REAL, 4=U16_REAL, 5=RAW
[-b biast]: Set Bias Tee, 1=enabled, 0=disabled(default)
[-v vga_gain]: Set VGA/IF gain, 0-15 (default 5)
[-m mixer_gain]: Set Mixer gain, 0-15 (default 5)
[-l lna_gain]: Set LNA gain, 0-14 (default 1)
[-g linearity_gain]: Set linearity simplified gain, 0-21
[-h sensivity_gain]: Set sensitivity simplified gain, 0-21
[-n num_samples]: Number of samples to transfer (default is unlimited)
[-d]: Verbose mode

-- Andrew VK4TEC --

darksidelemm, 2018-02-11:
Support for other SDRs will come much later on, probably though some kind of GNURadio Wrapper, so we can support whatever SDRs GNURadio can support.

Right now there is no 'drop-in' equivalent of rtl_fm and rtl_power for the AirSpy, and so adding support would require essentially a complete re-write of rtl_fm and rtl_power functionality, which is not something I have either the time or the inclination to do at the moment, given the RTLSDRs perform quite adequately in this application.

darksidelemm, 2018-02-11:
This looks promising: https://github.com/rxseger/rx_tools

I'll look at performing some tests on the output of those utilities and see how they compare to the librtlsdr utilities. If they check out OK (i.e. no performance degradation), then we'll switch this repository over to using them.

darksidelemm, 2018-02-12:
Have been playing with rx_tools in a separate branch (https://github.com/darksidelemm/radiosonde_auto_rx/tree/rx_tools). While the SoapySDR backend stuff is promising, the rx_power utility is giving some very very odd results running using my Airspy. (Works with with a RTLSDR!) 
The main issue is a massive frequency offset, which ppm corrections do not fix. This suggests some bug in rx_tools, and may be related to https://github.com/rxseger/rx_tools/issues/56

Until this is fixed, this issue will be on-hold. I may merge the changes i've made into master with the comment that all the rx_tools input stuff is HIGHLY experimental and broken.

closed 2022-03-14 by darksidelemm