projecthorus/radiosonde_auto_rx #39: Text file dumpto log file for kml preparation ( to generate a kml )

vk4tec, 2018-02-15:
Enhancement Request: 

I would like to create and append to a text file whilst the RadioSonde is being received 

If possible the filename would be called "<id>.log" where is the radiosonde id N123456.log for example

The log file would contain N123456.log :- 

date,time,id,lat,lon,speed, ( any other fields available ) 
date,time,id,lat,lon,speed, ( any other fields available ) 
date,time,id,lat,lon,speed, ( any other fields available ) 
date,time,id,lat,lon,speed, ( any other fields available ) 
date,time,id,lat,lon,speed, ( any other fields available ) 

The idea being that a user can use perl or phython to scan for /auto_rx/*.log 

And parse and create kml files. As needed.

For the purpose of 

1. Displaying high res tracks of the radio sonde path ( as APRS is only sent xx seconds ) 
2. For creating coverage maps in google earth 
3. For modelling antenna patterns based on location and signal strength 

Either hard coded or an option flag on auto_rx.py

Cheers

Andrew VK4TEC

vk4tec, 2018-02-15:
This is done

closed 2018-02-15 by vk4tec

darksidelemm, 2018-02-15:
Yes, only since this project started have we been logging data >_>

Point 3 is a bit difficult as there's no signal strength or SNR data available from the demodulator (and adding it would mean a whole re-write). 

Points 1 and 2 are simple enough. In fact I have a script which covers point 2 already (gets all the last recorded positions and dumps them into a KML file), which I'll look at cleaning up and adding to the utils directory.