projecthorus/radiosonde_auto_rx #42: Can we split the sonde logs and the system logs

vk4tec, 2018-02-17:
In preparing a parse file for generating kml from sonde files, i had to distinguish between system and sonde log files.

The only way I could to see to do this was a foreach on /log/*RS*.log

Can we go for these two directories for logging to seperate out the sonde / system logs 

/log/sonde
/log/system

This would seperate out the sonde log files from the system ones.

Enabling ease for future enhancements , by not having to filter system from sonde files.

Cheers

Andrew VK4TEC

darksidelemm, 2018-02-17:
Easy one to fix. I won't use separate folders, but i'll have either _system.log or _sonde.log at the end of the log file name. That's easy enough to deal with.

darksidelemm, 2018-02-17:
Fixed in https://github.com/projecthorus/radiosonde_auto_rx/commit/5da01c38a1346b36f05510bab500ec2d60d60697

closed 2018-02-17 by darksidelemm