projecthorus/radiosonde_auto_rx #50: prefix payload name to facilitate filetering on map

dbrooke, 2018-03-05:
If the habitat payload name/callsign was given a constant prefix (such as "RS_") then the sondes could be hidden on the map by using "!RS_*" in the search box

darksidelemm, 2018-03-05:
This is a good suggestion. I'm feeling pretty damn guilty about being responsible for flooding tracker.habhub.org with hundreds of sondes, and a way to filter them out has been on the cards for a while...

I'll try and get to this over the next few days - will make it such that all the auto-generated payload docs use a callsign prefixed with 'RS_'

darksidelemm, 2018-03-07:
Closed under https://github.com/projecthorus/radiosonde_auto_rx/commit/0495fa09cf7b4e858cb4d8c502a15267ce0b3cb6

closed 2018-03-07 by dbrooke