projecthorus/radiosonde_auto_rx #55: Check if Habitat Payload doc exists before creating one.

darksidelemm, 2018-04-01:
Currently exporting payload telemetry out of Habitat is difficult, because each new payload document created for a given callsign supersedes the previous one. As a result, you need to export telemetry from multiple payload IDs to obtain a complete flight.

It would be better if auto_rx checks to see if there is a payload document that matches the current callsign *before* creating a new document. This way we should end up with (hopefully) a single payload document, which can be used to export payload telemetry.

This is going to require digging through Habitat as to how we query payload documents, and then filter then by callsign.

darksidelemm, 2018-04-02:
It looks like habitat only supports exporting the entire list of payload documents:
https://habitat.readthedocs.io/en/latest/habitat/habitat/habitat/habitat.views.payload_configuration.html#habitat.views.payload_configuration.name_time_created_map
i.e. no filtering is possible, until you have the entire list. Not so useful if we are running on a slow internet connection.

An alternate approach would be to query spacenear.us for any recent positions for the current payload, using the URL:
http://spacenear.us/tracker/datanew.php?mode=1hour&type=positions&format=json&max_positions=10&position_id=0&vehicle=RS_<serial>
If *any* positions are returned, that indicates a valid payload document is present, and that one doesn't need to be created. A bit of a roundabout solution, but it would work.

darksidelemm, 2018-04-15:
Interim fix for this in https://github.com/projecthorus/radiosonde_auto_rx/pull/62

closed 2018-04-15 by darksidelemm

ED6E0F17, 2018-04-16:
I have already hit my quota of stupid mistakes for this week, so I am a bit nervous about mentioning this, but the fix seems to be disabled:

https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/habitat_utils.py#L152

````
 _callsign_present = False# check_callsign(serial)
````
I have been having issues with the habmodem app which I think is due to this issue, so I am happy that you are fixing it.

darksidelemm, 2018-04-16:
Derp! You are absolutely correct... Have fixed this in https://github.com/projecthorus/radiosonde_auto_rx/commit/8761adb1c082b660bc5ffcb2997874f67333c0dc

darksidelemm, 2018-04-16:
What is the issue you are encountering with 'habmodem'? (Is this the 'HAB Modem and Tracker' app for Android?)

ED6E0F17, 2018-04-24:
I am using Matt`s Android app (HAB Modem and Tracker) for "offline" mapping, but after a while it would stop fetching new positions. I think this was due to the payload docs changing when other Trackers started uploading sondes that I was already tracking. The wind from Camborne has not come this way since you fixed the payload doc issue, but I expect this to be OK now.

I also tested the new upload method with a broken internet connection, and it coped well with the disconnection.

(I did find another issue with the Habmodem app if the time wraps over midnight.)