projecthorus/radiosonde_auto_rx #60: Critical info captured in log files ( Habitat ID )

vk4tec, 2018-04-06:
Mark,

I am working on a HTML GUI , and making logs available via apache web server

Could you please add the HABITAT ID for example "VK4TEC_HOME" to system.log

INFO: Habitat Listener: Listener information <id> uploaded. Perhaps LAT and LON ?

My LAT LON
My Habitat ID
Sonde ID
Sonde Lat / Lon 
Sonde Height 
GPS Almanac attained
GPS Ephemeis attained

Example log files being cut at the moment

Soon to be replaced with MySQL HTML GUI

http://59.167.159.165/

// 

20180405-064906_system.log:2018-04-05 06:50:31,810 INFO:Habitat Listener: Created a payload document for RS_N1143473
20180405-065243_system.log:2018-04-05 06:54:01,016 INFO:Habitat Listener: Created a payload document for RS_N1143473
20180405-065639_system.log:2018-04-05 06:58:00,932 INFO:Habitat Listener: Created a payload document for RS_N1143473
20180405-121918_system.log:2018-04-05 17:09:01,662 INFO:Habitat Listener: Created a payload document for RS_N1033720
20180405-121918_system.log:2018-04-05 23:20:01,517 INFO:Habitat Listener: Created a payload document for RS_N1333230
20180406-114022_system.log:2018-04-06 11:40:25,166 INFO:Habitat Listener: Listener information uploaded.
20180406-114022_system.log:2018-04-06 17:10:58,978 INFO:Habitat Listener: Listener information uploaded.
20180406-114022_system.log:2018-04-06 17:11:31,106 INFO:Habitat Listener: Created a payload document for RS_N1033662
20180406-114022_system.log:2018-04-06 18:48:18,829 INFO:Habitat Listener: Listener information uploaded.
20180406-114022_system.log:2018-04-06 18:52:02,883 INFO:Habitat Listener: Listener information uploaded.
20180406-114022_system.log:2018-04-06 18:54:41,607 INFO:Habitat Listener: Listener information uploaded.
20180406-114022_system.log:2018-04-06 23:18:52,717 INFO:Habitat Listener: Listener information uploaded.
20180406-114022_system.log:2018-04-06 23:19:31,051 INFO:Habitat Listener: Created a payload document for RS_N1033651

vk4tec, 2018-04-06:
I think its important to keep sonde and system seperate, but pump system with as much info as possible

Andrew

darksidelemm, 2018-04-07:
The info you are after (the Habitat user callsign) is stored in the configuration file. Not a lot of sense duplicating that in the log file as well.
If you can read the log file you can also read the relevant bits from the configuration file just as well...

closed 2018-04-07 by darksidelemm