projecthorus/radiosonde_auto_rx #61: Temperature showing -273.0C

noengine, 2018-04-12:
How to get the actual meteorological data in the logs rather than just GPS coords and altitude

darksidelemm, 2018-04-12:
I'm assuming you are decoding a RS92.
Currently temperature decoding is only supported on the newer Vaisala RS41 radiosondes. Humidity decoding is not yet supported, but may be added in the future.

There is no plan to add support for the RS92s, since they are obsolete and should be phased out over the next year or so (depending on local stocks).

closed 2018-04-12 by darksidelemm

noengine, 2018-04-12:
Correct - RS92
I am interested in processing wx data directly from the sonde for Glider pilot weather briefings.
I'm stoked that you made auto_rx, btw! kudos!

darksidelemm, 2018-04-12:
You may want to try out SondeMonitor ( https://www.coaa.co.uk/sondemonitor.htm ), which supports decoding temperature/humidity data from RS92s, and will also provide a stuve diagram.
It's not free and not open source, but it might do what you're after until your local radiosonde launch site starts flying RS41s!