projecthorus/radiosonde_auto_rx #70: ImportError: No module named crcmod

pit711, 2018-06-17:
i get this error, any idea how to fix?

a@a:~/radiosonde_auto_rx/auto_rx$ python auto_rx.py
Traceback (most recent call last):
  File "auto_rx.py", line 22, in <module>
    from autorx.habitat import HabitatUploader
  File "/home/a/radiosonde_auto_rx/auto_rx/autorx/habitat.py", line 8, in <module>
    import crcmod
ImportError: No module named crcmod

Linux a 4.10.0-35-generic #39~16.04.1-Ubuntu SMP Wed Sep 13 09:02:42 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2

pit711, 2018-06-17:
sudo apt-getpython-crcmod

darksidelemm, 2018-06-17:
Only the second items in the software dependency list...
https://github.com/projecthorus/radiosonde_auto_rx/wiki#31-software-dependencies

pit711, 2018-06-18:
Issue was already closed,  Why did you comment anyway?

closed 2018-06-18 by pit711