projecthorus/radiosonde_auto_rx #79: Display old log to the web server page.

djecom1, 2018-07-07:
Add to the local web page the possibility to read the old log (text and / or cartography).
(if possible...!!!)

darksidelemm, 2018-07-07:
There is a log file to KML converter script here:
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/log_to_kml.py
Note: This utility requires the fastkml and shapely libraries, which can be installed using:

sudo pip install fastkml shapely

If you run it with the default options (i.e. python log_to_kml.py) it'll read in *all* the sonde log files, and output the all into one kml file (sondes.kml). Run the script with --help to see the command line options.

It this sufficient?

djecom1, 2018-07-07:
It's perfect ! Sorry, I didn't saw that script.. 
Thank you !

closed 2018-07-07 by djecom1