projecthorus/radiosonde_auto_rx #81: pip install of flask etc. requires python-setuptools

dbrooke, 2018-07-10:
Suggest adding python-setuptools to the apt-get install line in 3.1 of the wiki page.

Found during a fresh install on a 2018-06-27-raspbian-stretch-lite image so it's possible a full image includes python-setuptools but it shouldn't harm to add it anyway.

darksidelemm, 2018-07-11:
Hmm, i've always been installing from a raspbian-lite image without issues, but maybe something has changed. Have added it to the list.

closed 2018-07-11 by darksidelemm

dbrooke, 2018-07-11:
Thanks. Your comment made me take another look at why I had to add it manually. I deploy a base install using saltstack to all my Debian/Raspbian systems and as part of that my apt is configured to install only hard dependencies and not all recommended/suggested packages. So it's most likely due to that and wouldn't be necessary for most users for whom recommended packages would be automatically installed. Sorry for the noise!