projecthorus/radiosonde_auto_rx #85: Get data samples

joserrecio, 2018-08-02:
I´m trying to get the data samples on an array after receiving the signal whit airspy_rx. But I don´t know where is this array stored. Can anyone help me whit that?

darksidelemm, 2018-08-02:
Sorry, I'm not quite sure what you're trying to achieve here?

joserrecio, 2018-08-03:
I´m trying to get the samples of the received signal. When you store de signal in a wave file, this file is formed by some headers and data. I need to obtain the value of every sample of the received signal to demodulate it.

darksidelemm, 2018-10-08:
Is this still relevant? Otherwise I'm going to close this issue out.

Adding an 'option' to tee the received FM-demodulated signal to disk is still potentially useful, but I'm not sure it's quite what the OP is after...

darksidelemm, 2019-03-24:
Added ability to log demodulated audio to disk in https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.0.1
I consider this issue closed.

closed 2019-03-24 by darksidelemm