projecthorus/radiosonde_auto_rx #86: Web Interfaces 'times out' when not in foreground on Browser

darksidelemm, 2018-08-05:
The web interface appears to 'timeout' after some duration of being in a background tab of a web browser. 
This is probably related to the current 'short-polling' method being used for socket.io data transfers (as we don't have a proper Websockets-capable server in the backend). This needs to be confirmed however...

TheSkorm, 2018-08-08:
Haven't been able to replicate. Haven't seen an increase in memory over time. Will monitor it

DL2MF, 2018-10-23:
Can't confirm, I did use several different devices (laptop, tablet, smartphone) and operating systems (Linux, Windows, iOS, Android) and auto_rx / chasemapper were running > 24h in my local network without any issues to revoke a session after several hours.

lagidnyj, 2021-02-02:
can you tell me which parameter is responsible for clearing the web interface table? is it possible to increase the residence time of monitored probes to 12 or 24 hours? at least. I have probes flying at night, when I open the panel in the morning, I see nothing. then I have to go to the APRS and pick at the logs for the past 10 hours

darksidelemm, 2021-02-02:
That would be the 'archive age' setting of the [web] section of the config file: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L354

darksidelemm, 2021-04-18:
Giving up on this one. The web interface is functional enough as it is...

closed 2021-04-18 by darksidelemm