projecthorus/radiosonde_auto_rx #88: Add RS41 Burst-Kill timer support from upstream RS repo

darksidelemm, 2018-08-31:
Extraction of the burst-kill timer value has been added to the upstream RS repo as of https://github.com/rs1729/RS/commit/3e68a22786a3a8a376c64e785673c3f51459204b

- Add the timer value into the json blob we output from rs41demod
- Investigate adding the burst-timer counter value into the comment field when we create a Habitat sentence.

darksidelemm, 2019-03-15:
Found the countdown timer bytes in cal frame 0x32.
@rs1729 check out the changes I've made here: https://github.com/darksidelemm/radiosonde_auto_rx/blob/testing/demod/rs41dm_dft.c#L901

The countdown timer value is 0xFFFF until after burst, and then it counts down starting at the burst timer value in frame 0x32.

Currently doing some tests with it in my testing branch before pushing it into the main testing branch for wider use.

rs1729, 2019-03-15:
You mean the bt in frame 0x31?
Yes, after burst the first 2 bytes in 0x32 count down from there, good observation.
The burst detection takes a few hundred meters I guess, before it starts.
I think I noted somewhere that the bytes in 0x32 are not constant. Maybe that is the "RAM" and 0x00-0x31 is the "ROM".

darksidelemm, 2019-03-16:
I've added in just the countdown timer for now, in https://github.com/projecthorus/radiosonde_auto_rx/pull/140

closed 2019-03-16 by darksidelemm