projecthorus/radiosonde_auto_rx #98: HTTP output in log

SA7BNT, 2018-12-25:
Merry Christmas @all 

im getting this output in my log file. Is that normal or is it a issue on my side??
[image]

darksidelemm, 2018-12-25:
Hmm, you definitely shouldn't be seeing any output from the web-server, we should be inhibit messages from it (apart from errors) with these few lines: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/auto_rx.py#L397

What do you see right when auto_rx starts up? It should give some output regarding the flask webserver that might shed some more light on the situation.

SA7BNT, 2018-12-25:
Hey darksidelemm, 
here is the output.
[image]

And this is coming up when im open the browser
[image]

Viproz, 2018-12-25:
I can confirm the issue : https://pastebin.com/tj2ia39X
Theses logs were obtained after a reload of the webpage

darksidelemm, 2018-12-25:
OK, so Viproz has confirmed the messages are coming from geventwebsocket, which suggests something has pulled in gevent as a dependency. I'm not sure what at this point.

I've only briefly tried running the web interface using gevent as the webserver without much success. That it's working at all is pretty interesting. 

Noting the heap of crap in your log files, could you try adding the following after the block of 'logging.getLogger' lines at line 403 of auto_rx.py :

darksidelemm, 2018-12-25:
It's also possible to force the use of werkzeug as the server by instantiating the SocketIO object with the async_mode argument of 'threading'.

i.e. change the line https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/web.py#L40
to:

socketio = SocketIO(app, async_mode='threading')

This is not an ideal solution - gevent is probably a better server to use, though it would be a suitable workaround. The werkzeug server is sufficient for the very small number of concurrent users expected to access the auto_rx web interface.

darksidelemm, 2018-12-26:
OK, I'll do an update today to inhibit the gevent messages. I'l keep this issue open so we can track how things go with gevent. Still would like to know what pulled gevent in as a dependency...

darksidelemm, 2018-12-27:
Whoops, that shouldn't have been closed. Will re-open this until we figure out if gevent is usable.

closed 2021-04-18 by darksidelemm