projecthorus/radiosonde_auto_rx #105: Integration with Balena.io - fleet management RPi

SloMusti, 2019-01-27:
To simplify the deployment of the receiver and enable a more straightforward deployment, I have set-up a prototype docker deployment that can be used with http://balena.io system to mange a fleet of devices.

To do so, there were minimal changes required and it would be great to add support to the master that supports this. The changes made: https://github.com/SloMusti/radiosonde_auto_rx/commits/master
This in particular is just changing the default log path as `./log/` is not available when running the system in Balena, but must be placed on the data partition, thus `/data/log/` works. I suggest the log path should be configurable in settings.

The other required change is to make sure the config file is also on path `/data/` where it can be user modified.

To get this running on Balena.io system and deploy a fleet, I have published the code here: https://github.com/SloMusti/multicontainer-lora-radiosonde with the following key features:

darksidelemm, 2019-01-27:
Hi! 
Thanks for your work on this, this is amazing! Easy deployment of auto_rx has always been one of the things we needed to improve, and it looks like you're pretty much got the problem solved!

I have no issues with the log file directory being a configuration file parameter, in fact thats something I have been thinking about for a while. This would require changes to autorx/config.py to read in the path, and the other necessary changes where log files are written (in auto_rx.py and also in autorx/logger.py)

As for the configuration file, this can be specified with the --config option as it looks like you've found already. Is this sufficient?

Cheers,
Mark

SloMusti, 2019-01-27:
Sure `--config` is suitable, best would probably be to have also the  `--log` option that overrides the log folder as that would probably keep the code backwards-compatible for existing users.

Also, it is quite easy to add the following to the code: https://github.com/balena-io-projects/balena-dash so the receiver can have a nice display showing current results.

darksidelemm, 2019-01-27:
OK, breaking out out the log directory setting to --log is a good solution.

What changes did you have in mind for the dash support? I'd rather not add too much solution-specific stuff to the main auto_rx code if you get my meaning. Rather, i'd prefer to add suitable interfaces to enable other displays to be written. We have some of this already with the UDP output option, which is used by chasemapper.

SloMusti, 2019-01-27:
For the dash no changes are required for the main code, it can simply display auto_rx web interface.

darksidelemm, 2019-01-27:
OK, no problems. Will you be able to make the log directory option changes?

Also, I hang out on #highaltitude on Freenode IRC if you want to discuss further.

SloMusti, 2019-01-27:
Also have updated the https://github.com/SloMusti/multicontainer-lora-radiosonde configuration to work with this change, thus the system is now ready for testing. I kindly ask you or anyone else int eh community to go through this and ask if there are any problems or unknowns. The last step then remains a good tutorial and deployment of more base-stations should be simple.

darksidelemm, 2019-01-31:
Resolved as of https://github.com/projecthorus/radiosonde_auto_rx/commit/716af84b685cf423e8cd3b0dbd95abe9db17407d

closed 2019-01-31 by darksidelemm