projecthorus/radiosonde_auto_rx #109: Regex not matching with DFM

pit711, 2019-01-31:
Have some problems with this:
_WARNING:Payload ID DFMxx-xxxxxx does not match regex. Discarding._

2 bin files of the DFMs that cause this issues, created with:
### rtl_fm -p 0 -M fm -F9 -s 15k -f 405610000 > dfm_1.bin

https://mega.nz/#F!Epc3lQQA!UEkF4t7Kbvj4e8aYLKqSQg

darksidelemm, 2019-01-31:
So I've run some quick checks over those files. I first tested if they detected correctly using rs_detect, and then tested how they decoded using dfm09ecc.

Detect: cat filename | sox -t raw -r 15k -e s -b 16 -c 1 - -r 48000 -b 8 -t wav - highpass 20 lowpass 2000 2>/dev/null | ./rs_detect

Decode: cat filename | sox -t raw -r 15k -e s -b 16 -c 1 - -r 48000 -b 8 -t wav - highpass 20 lowpass 2000 2>/dev/null | ./dfm09ecc -vv --ecc

dfm_1.bin:
- Detected as DFM using rs_detect
- Only decodes with the -i (invert) option on dfm09ecc (though it's not detected as being inverted...)
- Looks to be a DFM06.
- Frames 115 through 6 decoded (frame count wraps at 255)
- Frames 115-179 do not have a valid callsign.
- After frame 179 the callsign becomes valid, but it varies a lot. Not sure if this is because the DFMs don't have a CRC that is checked, of if I'm not running the decoder correctly.

test.bin:
- Detected as a -DFM using rs_Detect (i.e. inverted)
- Only decodes if the -i option is *not* used!
- Frames 2 through 252 decoded.
- No valid callsign is detected.

I suspect this may be an issue with DFM06s specifically, but I can't be sure. Perhaps @rs1729 can shed some light on this. It's quite possible I'm not using the decoder correctly, or that there's a better decoder that I haven't found yet.

rs1729, 2019-02-01:
Interesting, I believe these are test-sondes from Graw/Nuernberg, probably some kind of DFM-17.
test.bin has the callsign/ID in "channel" C,
dfm_1.bin has the callsign/ID in "channel" D.
I don't know much about the new DFM-17 or derivatives.

You can look for the ID e.g. with

./dfm06 --ecc -r dfm_1.wav | grep "^D"
DC01210 [OK]   584F53514C716 [OK]   4D644A5C54417 [OK] 
DCFD341 [OK]   7E31FBB012008 [OK]   0000029002040 [OK] 
DCxxxx0 and DCyyyy1 higher/lower 16 bit:
xxxx=0x0121, yyyy=0xFD34
0x0121FD34=19004724

./dfm06 --ecc -r -i test.wav | grep "^C"
CC01130 [OK]   7E31FBAD0B008 [OK]   0000028802040 [OK] 
CCA8D31 [OK]   098D5A0136B01 [OK]   1D975F4807B22 [OK] 
0x0113A8D3=18065619

The latest dfm06ptu also looks for the ID in "channel" C, maybe I should look for D as well. But I don't know for sure if they are both DFM-17. I have seen a pilotsonde with ID in "channel" 7, but don't know much about these either.
DFM-09 has the ID in "channel A". Second nibble seems to be C for A, C, D.

I did not put the new IDs in dfm09dm_dft, yet.

[DFM-log.txt](https://github.com/projecthorus/radiosonde_auto_rx/files/2819626/DFM-log.txt)

rs1729, 2019-02-01:
Here I have updates
https://github.com/rs1729/RS/blob/master/demod/dfm09dm_dft.c
https://github.com/rs1729/RS/blob/master/dfm/dfm06ptu.c

I think the ID/SN is always in the highest "channel" (except for old DFM-06 with less then 6 "channels"). So in principle you could read more frames and avoid some DFM-06 false positives.
The polarity changed when there were DFM-09 with stm32, I believe. There are older DFM-09 with DFM-06 chips, and they transmit like DFM-06. The pilotsonde PS-15 that I know is same way as DFM-06 (and test sondes could always differ or even have scrambled frame structure).

darksidelemm, 2019-02-01:
At the moment I'm only using the dfm09ptu decoder in auto_rx, as rs_detect doesn't differentiate between DFM09s and DFM06s.
I'll look at merging the changes into the auto_rx repository today sometime (after changing this mornings sonde!)

rs1729, 2019-02-01:
DFM09 = -DFM06
rs_detect-sign is the same as dfm06ptu-sign, dfm09 has the opposite sign/polarity (except the dfm09 with dfm06-chips mentioned above).
However which sign depends on your setup (receiver/sdr-prog and sound card, when you decode audio). With rtl_fm I think the dfm06 should have positive sign.
Therefore in dfm09dm_dft I flipped the sign.
The dfm06ptu has an auto-option, that can detect the polarity. If you have a low threshold it could false detect in the same way as rs_detect, the flipped headers are not so different. Would be more reliable if several headers are detected. dft_detect gives correlation-magnitude. Then, decoding without errors will give confidence. And the ID of dfm06 is different from the 2-part-ID of dfm09.

darksidelemm, 2019-02-01:
OK... well at least with the auto_rx setup the receive polarity is going to be consistent across all stations, as they will be running the same FM demodulator code...

Annoyingly I don't have a DFM06 to test with, only a DFM09. Just so I have it all straight, could you re-confirm what I should be running for each detection output?
i.e. 
rs_detect DFM => dfm09dm_dft -i
rs_detect -DFM -> dfm09dm_dft 
??

And just to confirm, does dfm09dm_dft decode DFM06s correctly? Or is this where the issue where the ID is not found appears?

rs1729, 2019-02-02:
Yes, rs_detect <-> dfm06, hence -rs_detect <-> dfm09.
"dfm09dm_dft" is like "dfm06ptu -i -b2", only the header-detection is better (correlation), and since the dfm06 is getting old, I made the dft-version with dfm09-polarity.
Now after the update the ID-detection is also the same.

The DFMs all use the same modulation (mod polarity) and frame structure. The DFM-09 have more data.

Probably the new DFM-17 will transmit like the DFM-09, although the other test-sonde was inverted.

Actually you could decode both together (the header detection can give you the sign/polarity) as it is done for the M10 (the differential manchester code there changes sign from frame to frame). But I didn't want to lose the polarity-information for DFM-06/09.

rs1729, 2019-02-02:
Testing auto-polarity dfm09dm_dft. Perhaps that would be easier to use. But on weak signals it might drop some frames.

Just took a quick look on your json-modifications:
The new DFM-ID is usually 8 digits long (starting with YY I believe).
6 was for the DFM-06 and I took that as a minimum. But if you write the ID into sonde_id[], you should declare more space. Sorry that the ID-output is misleading.

I think the DFM with ID in "channel" 7 should be a pilotsonde (they don't have many sensors), so it is more a PS-15 than DFM-15. I'm not sure if there are different models. Don't know if you want to use different names, maybe the "DFM" is better for putting them in the Graw-category. I have not seen a PS-15 in real, they are so small.

darksidelemm, 2019-02-02:
OK, will fix...

darksidelemm, 2019-02-02:
OK, I've updated both the file in the RS repo PR, and the file in the auto_rx repo to handle 8-digit serial numbers.
I figure i'll leave all Graw devices as DFMxx for now, until we actually see some of them show up on the auto_rx network, then perhaps we can make some changes :-)

pit711, 2019-02-02:
its decoding my DFM06 and 09 now, good work, thx.
DFM09 is detected as DFM06

[https://github.com/pit711/radiosonde_auto_rx/blob/master/2019-02-02%2013_47_30-Radiosonde%20Auto-RX%20Status.png](url)

rs1729, 2019-02-02:
If it is a DFM-09 with DFM-06 inside (call it DFM-06/09), then it has the 6-digit ID and same polarity as DFM-06.
The early (real/stm32) DFM-09 with "channels" 0..A also had 6-digit IDs, but in ACxxxx and the DFM-09-polarity.
At maybe there are exceptions to this, the DFM-09-IDs are a bit confusing.

darksidelemm, 2019-02-02:
Looks like this one might be solved - yay! 
I guess I'll roll it out in the next major update (when Viproz's M10 changes get added in too) and we'll see how many more DFM06s start showing up on the map.
I'll close this issue now. If you encounter other issues, feel free to raise the issue again.

closed 2019-02-02 by darksidelemm