projecthorus/radiosonde_auto_rx #118: Frames are not accepted / dropped on Radiosondy.info

pit711, 2019-02-12:
I can decode DFM09 and M10 sondes successful but most time Radiosondy do not accept the APRS frames.
Are the frames in the right format? I asked the admin but he could not help me

[image]
[image]

darksidelemm, 2019-02-12:
Yep, it would be good to find out what radiosondy.info is expecting in terms of object name and comment format... without that we are kind of guessing. 
I've seen at least 2 different DFM callsign formats, and I thought we had already met one of them... but I guess not?

pit711, 2019-02-12:
his image for the raspberry is based on dxlAPRS, so it should be 100% in that format:
https://github.com/oe5hpm/dxlAPRS

darksidelemm, 2019-02-12:
If someone can dig through the dxlAPRS code and figure out how the serial numbers are converted into callsigns that would be appreciated. Once it's documented we should be able to implement it.

rs1729, 2019-02-12:
Is it the same ID that is used on their website?
https://radiosondy.info/sonde_archive.php?sondenumber=DF6396B3
It is a DFM-09 with serial number 18060979.
18060979 = 0x11396B3,
i.e. DF6xxxxx, with last 5 hex-nibbles (could work with DFM-06, though the DFM-06 hex-nibbles are the numbers on the radiosonde).
And then there are also
DF9yyyyyy
https://radiosondy.info/sonde_archive.php?sondenumber=DF9027722
Again it is a DFM-09, this time 17027722 (and 0x103D28A in data), so this time the hex in the data is converted to base10 and (the year?) chopped?

Maybe the DFM type is not identified properly.

pit711, 2019-02-13:
These M10s are also not accepted by Sondy.

[image]

darksidelemm, 2019-02-13:
Do you have some examples of the uploaded APRS sentences? (should be in the system log)

pit711, 2019-02-14:
Both not displayed on Radiosondy

`19-02-13 12:49:26,841 INFO:Detected new M10 sonde on 401.000 MHz!
2019-02-13 12:49:26,843 INFO:Halting Scanner to decode detected radiosonde.
2019-02-13 12:49:26,844 INFO:Scanner #00000003 - Waiting for current scan to finish...
2019-02-13 12:49:41,694 INFO:Scanner #00000003 - Scanner Thread Closed.
2019-02-13 12:49:41,696 INFO:SDR #00000003 has been allocated to Decoder (-M10, 401.000 MHz).
2019-02-13 12:49:44,070 INFO:Decoder #00000003 M10 401.000 - Starting decoder subprocess.
2019-02-13 12:50:15,139 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724805 *125028h4825.59N/00250.13EO108/040/A=094562 Clb=6.1m/s t=-57.9C 401.000 MHz Type=M10 https://radiosondy.info/ !w-!!

2019-02-13 12:51:12,688 ERROR:Decoder #00000003 M10 401.000 - RX Timed out.
2019-02-13 12:51:13,691 INFO:Decoder #00000003 M10 401.000 - Closed decoder subprocess.
2019-02-13 12:51:14,146 INFO:SDR #00000003 has been allocated to Scanner.
2019-02-13 12:51:16,560 INFO:Scanner #00000003 - Starting Scanner Thread
2019-02-13 12:51:16,562 INFO:Scanner #00000003 - Running frequency scan.
2019-02-13 12:51:19,408 INFO:Scanner #00000003 - Detected peaks on 8 frequencies (MHz): [ 402.71  403.4   401.    405.    400.99  400.6   402.    401.01]
2019-02-13 12:51:32,581 INFO:Detected new M10 sonde on 401.000 MHz!
2019-02-13 12:51:32,582 INFO:Halting Scanner to decode detected radiosonde.
2019-02-13 12:51:32,583 INFO:Scanner #00000003 - Waiting for current scan to finish...
2019-02-13 12:51:46,551 INFO:Scanner #00000003 - Scanner Thread Closed.
2019-02-13 12:51:46,553 INFO:SDR #00000003 has been allocated to Decoder (M10, 401.000 MHz).
2019-02-13 12:51:48,917 INFO:Decoder #00000003 M10 401.000 - Starting decoder subprocess.
2019-02-13 12:52:40,233 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724805 *125253h4825.05N/00252.89EO117/055/A=097474 Clb=6.1m/s t=-56.6C 401.000 MHz Type=M10 https://radiosondy.info/ !wE3!

2019-02-13 12:53:20,160 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724805 *125326h4824.86N/00253.65EO098/068/A=098142 Clb=6.4m/s t=-55.5C 401.000 MHz Type=M10 https://radiosondy.info/ !wYA!

2019-02-13 12:54:10,130 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724805 *125423h4824.52N/00255.18EO092/064/A=099190 Clb=5.0m/s t=-53.5C 401.000 MHz Type=M10 https://radiosondy.info/ !wA/!

2019-02-13 12:55:09,182 ERROR:Decoder #00000003 M10 401.000 - RX Timed out.
2019-02-13 12:55:10,185 INFO:Decoder #00000003 M10 401.000 - Closed decoder subprocess.
2019-02-13 12:55:11,109 INFO:SDR #00000003 has been allocated to Scanner.

`

`019-02-12 01:00:29,327 INFO:Detected new M10 sonde on 401.000 MHz!
2019-02-12 01:00:29,328 INFO:Halting Scanner to decode detected radiosonde.
2019-02-12 01:00:29,329 INFO:Scanner #00000003 - Waiting for current scan to finish...
2019-02-12 01:00:42,982 INFO:Scanner #00000003 - Scanner Thread Closed.
2019-02-12 01:00:42,984 INFO:SDR #00000003 has been allocated to Decoder (M10, 401.000 MHz).
2019-02-12 01:00:45,356 INFO:Decoder #00000003 M10 401.000 - Starting decoder subprocess.
2019-02-12 01:01:20,122 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724037 *010133h4744.45N/00310.24EO090/115/A=109793 Clb=4.8m/s t=-27.3C 401.000 MHz Type=M10 Radiosonde http://bit.ly/2Bj4Sfk !w%5!

2019-02-12 01:01:55,152 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724037 *010203h4744.31N/00311.51EO086/109/A=110366 Clb=7.2m/s t=-25.5C 401.000 MHz Type=M10 Radiosonde http://bit.ly/2Bj4Sfk !ws1!

2019-02-12 01:02:25,196 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724037 *010236h4744.11N/00313.04EO098/136/A=111099 Clb=0.2m/s t=-24.2C 401.000 MHz Type=M10 Radiosonde http://bit.ly/2Bj4Sfk !w]_!

2019-02-12 01:03:00,187 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724037 *010312h4743.92N/00314.82EO104/151/A=111842 Clb=4.0m/s t=-23.2C 401.000 MHz Type=M10 Radiosonde http://bit.ly/2Bj4Sfk !w[K!

2019-02-12 01:03:57,374 ERROR:Decoder #00000003 M10 401.000 - RX Timed out.
2019-02-12 01:03:58,377 INFO:Decoder #00000003 M10 401.000 - Closed decoder subprocess.
2019-02-12 01:03:59,529 INFO:SDR #00000003 has been allocated to Scanner.
2019-02-12 01:04:01,892 INFO:Scanner #00000003 - Starting Scanner Thread
2019-02-12 01:04:01,894 INFO:Scanner #00000003 - Running frequency scan.
2019-02-12 01:04:04,294 INFO:Scanner #00000003 - Detected peaks on 3 frequencies (MHz): [ 401.    404.99  401.01]
2019-02-12 01:04:07,899 INFO:Detected new M10 sonde on 401.000 MHz!
2019-02-12 01:04:07,900 INFO:Halting Scanner to decode detected radiosonde.
2019-02-12 01:04:07,901 INFO:Scanner #00000003 - Waiting for current scan to finish...
2019-02-12 01:04:22,654 INFO:Scanner #00000003 - Scanner Thread Closed.
2019-02-12 01:04:22,656 INFO:SDR #00000003 has been allocated to Decoder (-M10, 401.000 MHz).
2019-02-12 01:04:25,048 INFO:Decoder #00000003 M10 401.000 - Starting decoder subprocess.
2019-02-12 01:05:00,097 ERROR:APRS-IS - Error converting telemetry to APRS packet - 'ascii' codec can't decode byte 0x81 in position 2: ordinal not in range(128)
[... cut ...]
2019-02-12 01:07:00,214 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724037 *010711h4744.00N/00327.06EO069/128/A=115742 Clb=-26.4m/s t=-15.5C 401.000 MHz Type=M10 Radiosonde http://bit.ly/2Bj4Sfk !w=g!

2019-02-12 01:07:35,155 INFO:APRS-IS - Uploaded to APRS-IS: PIRAT1-14>APRARX,SONDEGATE,TCPIP,qAR,PIRAT1-14:;ME724037 *010746h4744.03N/00328.83EO101/109/A=109589 Clb=-41.8m/s t=-26.6C 401.000 MHz Type=M10 Radiosonde http://bit.ly/2Bj4Sfk !wCK!

2019-02-12 01:08:30,735 ERROR:Decoder #00000003 M10 401.000 - RX Timed out.
2019-02-12 01:08:31,738 INFO:Decoder #00000003 M10 401.000 - Closed decoder subprocess.`

Viproz, 2019-02-14:
I can confirm this issue, I had the same problem with M10s, I could see in the log that it uploaded to APRS but it wasn't showing up on the map.
I think the title is misleading as this issue is not related to the ID of the RS (for APRS with the M10 we're uploading the dxlID generated by the M10 decoder which is the same as the ID generated by dxl code, tested) but an issue with the upload to APRS.

Viproz, 2019-02-14:
Just one precision @pit711  since I've seen some of your Paris' decoded M10 position on APRS but not on radiosondy, you are using this in your config file right ?
`# APRS-IS server to upload to.
# Default to radiosondy.info for now, to allow stats to show up on http://radiosondy.info
# Packets are forwarded onto the rest of the APRS-IS network from radiosondy.info.
# If you wish to inject packets directly into the APRS-IS network, use rotate.aprs2.net
aprs_server = radiosondy.info`

pit711, 2019-02-14:
yes i do, i will change to non-forwarding server.

Viproz, 2019-02-16:
No no sorry this is not what I meant, using the default setting is good, the good config in this case would be the one I posted, I just wanted to check that you hadn't changed it. The only callsign issue there is is with DFMs, not with M10s.

darksidelemm, 2019-07-20:
Closed in https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.1.3.2

closed 2019-07-20 by darksidelemm