projecthorus/radiosonde_auto_rx #119: RS41 aux data

TheSkorm, 2019-02-13:
We should be logging and reporting aux data for RS41s

TheSkorm, 2019-02-28:
Some samples recorded for testing

rs1729, 2019-03-03:
If there are more than 1 aux-string in one frame, it overwrites the string with the last aux-message, however if a previous one was longer, the leftovers are still visible. (also the last character is chopped.)

rs41dm_iq.c was prepared for aux and conf data. I added json-output for aux-data, so you can compare.

TheSkorm, 2019-03-04:
Nice catch @rs1729 ; My real world sample only had one sensor so didn't have this issue. I'll patch this later. It should be as simple as adding a null char after the loop. I'm not sure how we should handle multiple aux frames though. I guess we could make the JSON have an array of aux data but I'm not sure how we'd handle that with the logging

rs1729, 2019-03-04:
Here
https://github.com/rs1729/RS/blob/master/demod/rs41dm_iq.c
I concatenate the aux-strings, using '#' as separator.

darksidelemm, 2019-03-13:
Resolved by https://github.com/projecthorus/radiosonde_auto_rx/pull/138

PR #136 (2019-03-10): Minor tweaks to auxdata collection
Fixes #119
Tested against ozonesonde sample and normal rs41 sample
closed 2019-03-13 by darksidelemm