projecthorus/radiosonde_auto_rx #132: Is m2k2 supported?

runahue, 2019-03-04:
Hello friends:
I am Ruben EA5BZ.
A little question: Is m2k2 sonde supported?
It is the one used in my zone.

Thank's a lot.

Viproz, 2019-03-04:
No, I should add them

runahue, 2019-03-04:
Thank's for your answer.
It would be great for us.

Viproz, 2019-03-04:
Actually it's going to be harder than just a small change, still want to do it but I'm going to need a few samples and a few things, can you join us on #highaltitude on freenode ? (http://webchat.freenode.net/?channels=highaltitude)

darksidelemm, 2019-03-05:
Discussion in #highaltitude ended up showing that the radiosonde signal was actually a M10. 
I'm a bit hesitant about adding support for very old / obsolete radiosonde models, since it will be a lot of effort for very little gain.

runahue, 2019-03-05:
It is running fine.
But it is logging only position.
Can we log Meteo Information from Sonde?

2019-03-05T11:51:39Z,M10-T-806-2-12346,1551786699,37.99124,-0.74471,9936.1,-49.2,M10,403.400
2019-03-05T11:51:40Z,M10-T-806-2-12346,1551786700,37.99126,-0.74442,9940.4,-49.2,M10,403.400
2019-03-05T11:51:41Z,M10-T-806-2-12346,1551786701,37.99130,-0.74413,9944.8,-49.3,M10,403.400
2019-03-05T11:51:43Z,M10-T-806-2-12346,1551786703,37.99136,-0.74358,9953.8,-49.3,M10,403.400
2019-03-05T11:51:44Z,M10-T-806-2-12346,1551786704,37.99138,-0.74331,9957.9,-49.4,M10,403.400
2019-03-05T11:51:46Z,M10-T-806-2-12346,1551786706,37.99141,-0.74278,9966.8,-49.4,M10,403.400
2019-03-05T11:51:47Z,M10-T-806-2-12346,1551786707,37.99143,-0.74251,9971.3,-49.5,M10,403.400
2019-03-05T11:51:48Z,M10-T-806-2-12346,1551786708,37.99145,-0.74224,9976.1,-49.5,M10,403.400
2019-03-05T11:51:49Z,M10-T-806-2-12346,1551786709,37.99148,-0.74196,9980.7,-49.5,M10,403.400
2019-03-05T11:51:50Z,M10-T-806-2-12346,1551786710,37.99152,-0.74168,9985.2,-49.6,M10,403.400

2019-03-05 12:35:52,104 INFO:Reading configuration file...
2019-03-05 12:35:54,192 INFO:Config - Tested SDR #0 OK
2019-03-05 12:35:54,199 INFO:Started Flask server on http://0.0.0.0:8089
2019-03-05 12:35:54,204 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2019-03-05 12:35:54,209 INFO:SDR #0 has been allocated to Scanner.
2019-03-05 12:35:55,034 INFO:RTLSDR - Attempting to reset: Bus: 1  Device: 79
2019-03-05 12:35:59,108 INFO:Scanner #0 - Starting Scanner Thread
2019-03-05 12:35:59,113 INFO:Scanner #0 - Running frequency scan.
2019-03-05 12:36:19,173 INFO:Scanner #0 - Detected peaks on 3 frequencies (MHz): [ 403.4   403.41  403.39]
2019-03-05 12:36:23,159 INFO:Detected new M10 sonde on 403.400 MHz!
2019-03-05 12:36:23,161 INFO:Halting Scanner to decode detected radiosonde.
2019-03-05 12:36:23,163 INFO:Scanner #0 - Waiting for current scan to finish...
2019-03-05 12:36:36,752 INFO:Scanner #0 - Scanner Thread Closed.
2019-03-05 12:36:36,754 INFO:SDR #0 has been allocated to Decoder (M10, 403.400 MHz).
2019-03-05 12:36:38,879 INFO:Decoder #0 M10 403.400 - Starting decoder subprocess.
2019-03-05 12:36:56,293 INFO:Telemetry Logger - Opening new log file: /home/pi/radiosonde_auto_rx/auto_rx/log/20190305-113656_M10-T-806-2-12346_M10_403400_sonde.log
2019-03-05 12:38:42,446 INFO:Telemetry Logger - Closed log file for M10-T-806-2-12346
2019-03-05 12:38:48,956 INFO:Telemetry Logger - Using existing log file: /home/pi/radiosonde_auto_rx/auto_rx/log/20190305-113656_M10-T-806-2-12346_M10_403400_sonde.log

darksidelemm, 2019-03-11:
Closing this issue - decoding of humidity data is not planned, and is more an issue better raised on the upstream RS repository.

closed 2019-03-11 by darksidelemm