projecthorus/radiosonde_auto_rx #147: Support for U.S. radiosondes

jelson, 2019-03-19:
Hi! I just saw your talk at the Linux conference in Christchurch. At the end you mentioned there currently is not support for U.S. radiosondes. Is there any ongoing work to add support?

darksidelemm, 2019-03-19:
Depends on the radiosonde!

From what I understand, in the US many sites are currently using the 1680 MHz radiosonde band. Usage of this band poses challenges for an automatic RX station, including:
- Difficulty in building a well performing omni-directional receive station (most of the official ground stations will be using tracking antennas)
- Difficulty in tracking transmit frequency drift (Though there is some work progressing to help with this)

There is a transition activity to move away from the 1680 MHz band, back to the worldwide 400-406 MHz band - this will make things a lot easier. 

I am aware of 3 different radiosonde types currently in use in the US, these are:
- Vaisala RS92-NGP (1680 MHz version) - There is some experimental support for these in auto_rx now, but it's untested on real-world sondes. These sondes are obsolete, and are (hopefully) being phased out.
- Lockheed Martin LMS6 - No support for these at the moment.
- Intermet iMet-1 and iMet-4 - We have support for the 'narrow bandwidth' iMet sondes (iMet 4, and some versions of the iMet1) operating on the 400 MHz band, though this is very much in testing at the moment, and hasn't been tried on real-world sondes (only on the bench).

What I really need is more information on what is flying, so I can focus efforts appropriately.

jelson, 2019-03-19:
Thanks so much for the detailed response. Is the problem that the 1680 Mhz band is more attenuated by the atmosphere, and thus omnidirectional antennas, or receivers not corrected for frequency drift, can't pick it out of the noise floor?

Also, when you say "No support" for the LMS6, does that mean you can pick up the signal but there isn't yet support for decoding its packet format?

I live in Seattle, WA, and have on my todo list to go out to Quillayute, WA, where they launch sondes twice a day, and watch a launch. If it would help, I could ask them what they're flying and any other details that might be pertinent.

darksidelemm, 2019-03-20:
The issue with the 1680 MHz band is mainly one of path loss, yes. A lot of the 1680 MHz sonde chasers I've seen are using higher gain antennas like helicals or dishes. While auto_rx does support driving a rotator (to track the sonde), this does increase complexity a lot.

The drift issues are kind of a separate problem, which is possible to solve, but I just haven't gotten to it yet!

There is a LMS6 decoder available, but I really need to get hold of one to be sure the scanning & detection aspects of auto_rx are going to work properly. I guess I could add it in and hope it works...

Yes, more information on what they are launching, what frequency they use, and what their future plans with regards to sonde models and frequencies would be useful.

jelson, 2019-03-20:
FWIW, the [National Weather Service web site](https://www.weather.gov/upperair/factsheet) has the text I've pasted below. The RS92-NGP looks from the Vaisala web site like a 400 Mhz device so maybe we're in luck?

>NWS currently uses two types of GPS radiosondes:
>* [LMS-6](https://www.weather.gov/uppereair/LargeLMS6.htm)  built by Lockheed Martin
>* [RS92-NGP](https://www.weather.gov/upperair/LargeViasala.htm) built by Vaisala
>
>Note: The Mark IIA Radiosonde was phased out of the NWS weather balloon network in early 2015.

darksidelemm, 2019-03-20:
The RS92-NGP is the 1680 MHz version of the RS92. I'm guessing the LMS6's they are using are also the 1680 MHz versions, so the same issues.

I'll try and get my changes to help with frequency drift released soon (at least in the testing branch). It would be worth having a look on the 1680 MHz band to see if you can see anything - you will likely need to get a suitable antenna and a SAW/Preamp like this one: https://www.nooelec.com/store/sawbird-goes.html

jelson, 2019-03-20:
Thanks for the advice - no rush on getting stuff into testing, I'm probably at least a month or two from being able to do the experiment. But I'm excited to try it!

rs1729, 2019-03-20:
Added JSON output to
RS/demod/mod/lms6mod.c

https://www.nws.noaa.gov/os/notification/notifications.php
Here one can find launch sites that transitioned to lms6 (403MHz) or rs92-ngp (or rs41 or dfm09 or lms6-1680).
Looks like
Quillayute 72797 KUIL
https://www.nws.noaa.gov/os/notification/tin13-09vaisala_rs92cca.txt
transitioned to rs92-ngp.
The WMO Radiosonde Identification Code also indicates rs92-ngp (code 52)
http://weather.uwyo.edu/upperair/sounding.html
(though sometimes stations use wrong codes)

jelson, 2019-03-20:
Thanks @rs1729  - I'm trying to understand the relationship between this auto_rx project and your RS project; is RS a dependency of auto_rx? Or are they two separate implementations of the same basic idea?

rs1729, 2019-03-20:
auto_rx uses the decoders from rs1729/RS.
They did the auto_rx part, i.e. automated scanning for radiosondes and decoding, reporting to a server, wind/landing prediction, and so on. I don't know much about this part, @darksidelemm will fill in, but I try to do some adjustments to the decoders and detection-tools so they can be easier integrated in the auto_rx chain.

jelson, 2019-03-20:
Got it, thanks. It looks like the auto_rx code is a copy of RS, not a submodule - does that mean if I want to use your new decoder I need to clone auto_rx, clone RS and copy the files over?

rs1729, 2019-03-20:
No, I don't think that is recommended.
If there are changes to RS that improve decoding, @darksidelemm and @TheSkorm test it and include it then in their RS-fork as soon as possible. Maybe they can explain it in more detail.
At rs1729/RS I have different variants of the decoders, and although they took everything over to auto_rx, they use/update only a subset, only what is needed for auto_rx.
There is a testing branch that may have newer features, but they try to make sure, that the data that is sent to the sondehub server is consistent, so it can take a while before it is included in the master branch.

But there are not so many changes for the decoders. Probably there will be more changes to the signal processing and demodulation at baseband (getting the data from the rtl_sdr to the decoder).

TheSkorm, 2019-03-20:
Submodules is something we've been considering; we just need to make sure that all the changes we've made to the decoders end up upstream or write patch files as part of the build.

For the moment it's just been easier to include our own copy of them 

Sent from Nine

darksidelemm, 2019-03-23:
Im going to close this and open some tracking issues for RS92-NGP and LMS6 support.

closed 2019-03-23 by darksidelemm

rs1729, 2019-07-05:
WMO codes, soundings in May 2019.
11: LMS6-403MHz
17: DFM-09
23,24,25,41,42: RS41
35: iMS100
52: RS92-NGP (1680MHz)
77: M10
80,81: RS92
82: LMS6-1680MHz
[image]

darksidelemm, 2019-07-05:
Excellent information, thanks!!

rs1729, 2019-07-06:
Don't know about soundings in Cuba
https://www.loc.gov/law/help/us-treaties/bevans/b-cu-ust000006-1225.pdf

Here is a worldwide list, 2019/May:
station: sonde_type # lat lon
[all_stat_2019may.txt](https://github.com/projecthorus/radiosonde_auto_rx/files/3364607/all_stat_2019may.txt)
a few stations are missing

The WMO code are in the common code tables
http://www.wmo.int/pages/prog/www/WMOCodes/WMO306_vI2/LatestVERSION/LatestVERSION.html
Sometimes the reported WMO codes in the sounding data are wrong.

[image]

rs1729, 2019-07-07:
Maps 2018 and 2019:
#https://github.com/rs1729/RS/issues/15
E.g. France is not reporting in 2019.

rs1729, 2021-02-01:
New world map 2021/01:
https://github.com/rs1729/RS/issues/15#issuecomment-771213447

darksidelemm, 2021-02-01:
Very nice, thanks!

ka9q, 2023-11-11:
The NWS site at Miramar Marine Corps Air Station in San Diego is flying the Vaisala RS41-SGP (i.e., the model with the pressure sensor). We have recovered many.

ka9q, 2023-11-11:
Vandenberg Space Force Base seems to fly both the LMS6-403 and the LMS-403-2, often in flurries before Falcon 9 rocket launches. The LMS-403-2 sondes rise, pop and fall normally while the LMS6-403 sondes often become floaters at 16,500-17,500 meters (54,000-57,000 ft) and travel over Los Angeles and Orange or Riverside counties, apparently until battery exhaustion. We don't know if this is intentional, but it is very persistent.