projecthorus/radiosonde_auto_rx #152: Improved Peak Detection

darksidelemm, 2019-03-23:
The current peak detection approach is quite coarse and doesn't reliably detect the centre frequency of most radiosonde types. 
We currently get around this by quantising the detected peaks to 10 kHz boundaries. This works OK for all 400 MHz radiosondes we currently support, however it will break with sondes that drift considerably. It will also likely break on the wider bandwidth 1680 MHz radiosondes.

This could possibly be improved by using the known spectral masks of the various supported radiosonde types, and correlating these masks with the output of rtl_power (or some other utility that produces FFT data). 

Maybe a spectrogram could be produced for the ~20 second snapshot we currently capture, and computer-vision techniques be used?

Constraints would be:
- Needs to run on a Raspberry Pi 2/3 in a reasonable amount of time (maybe cap the processing at 5-10 seconds).

rs1729, 2019-03-23:
Do you filter the rtl_power output before you look for peaks?

For me the following works: (For signals with bandwidth about 6-12 kHz)
Bin size about 100 Hz. Taking the power-average over maybe 4 sec. (This would be done by rtl_power?)
To smooth the db-power values there are two simple possibilities:
- Average over 4800 Hz (i.e. about 48 bins). A moving average could be used to save some resources. Very narrow peaks could be filtered in advance.
- Since all radiosonde signals are symmetric, with the exception of M10, a kind of symmetric average could be taken, i.e. how much do the values to left of the peak correlate with the values to the right. Here I sum over a length of 4000 Hz in both directions.
This "average" gives more distinct peaks for radiosonde signals.

Now one would have to look for local maxima, above the local noise floor. Furthermore a peak usually comes with an increase of deviation from the local average. Maybe the filtered db_power-data would give more accurate peaks with your detect_peaks-function?

As example I put scan_fft_simple.c and plot_fft_simple.py in my RS/scan directory.
scan_fft_simple reads iq-wav files and puts the db-values in db2.txt that can be read and plotted by plot_fft_simple. scan_fft_simple can also read raw data:
scan_fft_simple - <sample_rate> <bits_per_sample> <file.bin>
bits_per_sample is either (unsigned) 8, (signed) 16, or (float) 32.

For sample rates much smaller than 1e6 the FFT size and bin size would have to be adjusted, otherwise the plot is a bit nervous. Though it is only to illustrate how smoothing could help peak detection.

So for other signal bandwidth and asymmetric M10 signals it would need additional considerations.
But for me it detects the signal and center frequency accurately.
But correlating the signal frequency spectrum is definitely a thing to try. Though the templates are not easily generated I guess, needs some kind of database.

darksidelemm, 2019-08-27:
Hi! 

I guess the issue at the moment is that dft_detect works off FM-demodulated audio. Frequency offset can be determined from DC bias in the demodulator output, but it will require careful control over the FM demodulator settings.

With the now not-so-experimental decoders, tracking 20 kHz of drift isn't that difficult. You just feed the FSK demodulator with a bit more bandwidth, and it will find and track the signal well. fsk_demod also provides the frequency estimation outputs, which gives the potential for the SDR to be re-tuned if the signal starts to drift towards the edge of the receiver passband. I believe newer versions of rtl_fm allow for command via a TCP interface, which would allow us to complete the control loop. 

Hopefully the LMS6 gets phased out, as it's a very very old design, and the drift it exhibits is probably the worst of all the sondes currently in use.

darksidelemm, 2019-08-27:
> The latest dft_detect update also works with IQ samples (--iq for centered
> IF, --IQ for baseband). I need lowpass filtering before FM-demodulation
> (FM-demod, so I don't need all the different modulation parameters for each
> radiosonde). At the same place, dc-ofs can be removed easily, tested with
> rs41 e.g., detection works good, only for M10 the carrier offset interferes
> when the FFT-window is too wide. And the detected offset would have to be
> reported/calculated in the output.
> But I will see how it works in general and if it is ok, I can put in a
> dc-option.
>
> For the new decoder that decodes multiple signals in baseband in parallel,
> I wanted to include some frequency correction. But I still don't know what
> would be best, how to update the decoder-queue.
> For most of the radiosondes it would be ok to scan every few minutes and
> restart the decoders, but for rs41 e.g. you don't want to lose calibration
> data.
> So for known signals at the same time it works now, but a
> scan&detect-thread that updates the decoder-queue would be fine.
> https://youtu.be/5YXP9LYUgLs
> It did work also with 6Msps airspy_rx, but I didn't test airspy much. Also
> don't know how many signals would work on a raspi.

That multi-decoder is very neat! I guess what would make it *really* useful
is being able to add/remove sonde decoders while it's running, perhaps via
some TCP command interface, and possibly update currently running
demodulators. That way if the scanner detects another sonde still within
the receiver passband, you could just add it to the already running
decoder. Would be a bit more difficult if the new sonde was outside the
*current* receiver bandwidth - this may require re-tuning of the SDR, and
hence updating any existing running demodulators.

rs1729, 2019-08-27:
The latest dft_detect update also works with IQ samples (--iq for centered IF, --IQ <fq> for baseband). I need lowpass filtering before FM-demodulation (FM-demod, so I don't need all the different modulation parameters for each radiosonde). At the same place, dc-ofs can be removed easily, tested with rs41 e.g., detection works good, only for M10 the carrier offset interferes when the FFT-window is too wide. And the detected offset would have to be reported/calculated in the output.
But I will see how it works in general and if it is ok, I can put in a dc-option.

For the new decoder that decodes multiple signals in baseband in parallel, I wanted to include some frequency correction. But I still don't know what would be best, how to update the decoder-queue.
For most of the radiosondes it would be ok to scan every few minutes and restart the decoders, but for rs41 e.g. you don't want to lose calibration data.
So for known signals at the same time it works now, but a scan&detect-thread that updates the decoder-queue would be fine.
https://youtu.be/5YXP9LYUgLs
It did work also with 6Msps airspy_rx, but I didn't test airspy much. Also don't know how many signals would work on a raspi.

rs1729, 2019-08-27:
Yes, that's probably what I meant by "adding to the decoder queue". What I was trying to say:
If a radiosonde signal drops below detection, the decoder thread should be stopped. And if the scanner detects a new radiosonde signal, a new thread should be started/added.

But I don't know if the baseband should be better fixed. However one could think about using the librtlsdr to control the baseband.
And I don't know how many radiosonde types I would include in the multi-decoder.

rs1729, 2019-08-29:
dft_detect --dc can now compensate a few kHz offset. For fm-decoding in can be done the same way in the decoders. But for iq-decoding I don't know if I do the estimate via fm-demodulation or change the correlation/header-detection.
@darksidelemm 
For a communication between decoder and scanner there are several possibilities. However if it is not integrated as modules in the same programme, the decoder can report its status, but since it reads the samples from stdin, the scanner cannot use this channel.
Can you elaborate on "TCP command interface" a bit? Would it need some additional socket-library?

darksidelemm, 2019-08-29:
What I had in mind was a single 'server' that communicates the SDR and accepts a fairly wide bandwidth IQ stream.
Clients make a connection via TCP to this process, and send some kind of command that specifies the frequency, bandwidth, and modulation they want. The server then provides a stream of samples down that TCP connection. Multiple clients can connect, and request different frequencies / bandwidths, etc. Clients can update the server with a different frequency if, for example, you want to track the centre frequency of a sonde. 
I suspect doing something like this could be fairly CPU intensive, and I'm not sure how well it would work on something like a Raspberry Pi. It's really a large project all of its own!

darksidelemm, 2019-08-29:
I guess the same server could also do a rtl_power like thing where it either does a FFT of the bandwidth it's looking at, or perhaps does a few quick hops to cover a wider bandwidth. Definitely a fairly complex project.
Targeting *just* IQ output would probably make it simpler (each channel would just be a mixer and then a filter/decimation step).

miweber67, 2019-08-31:
It seems like something that does that is already available... The aisrpy spyserver fronts their own hardware but also the RTLSDR dongles. It supports multiple clients with different DDC chains, sending them IQ data in one of a few specified formats, and optionally FFT data for the same slice.

There is client code in the gqrx project, and I think also in the osmocomm library.

So, it seems that all one would need to do is write a replacement for rtl_fm that provides the desired IQ samples at the desired rate and pipe them in to whatever can FM demod them and pass them on to the rest of the usual chain of processors.

Seems that would work?

darksidelemm, 2019-08-31:
Yep, that might be a viable solution! Unfortunately I don't really have the
time to work on it much..
I guess what would be needed (at least to make it work as a 'drop-in'
replacement right now), is a command-line client that connects to the
backend, and produces the required output on stdout.

miweber67, 2019-08-31:
OK, perhaps I can add this to my list of projects to poke at... I would need to understand what the required output is, though. Is it just the 8-bit IQ stream? Into which component would that be fed?

darksidelemm, 2019-09-01:
As a start, a 'drop-in' replacement for rtl_fm would be good  (noting that we still use rtl_power to get spectrum data, but this can be looked at later).

The decoder commands are generated in this function: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/decode.py#L405

As an example, for the RS41: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/decode.py#L436

It's a bit difficult to read there, but essentially it's calling rtl_fm with a few arguments to set it up into raw IQ output mode (signed 16-bit ints), which in the case of the 'experimental' decode chains is passed into fsk_demod (but could also be passed into @rs1729's decoders in the same format). 
So having something that emulates rtl_fm, and can handle similar arguments (or at least ignore the ones that it doesn't use) would be a starting point.
An example command would be: rtl_fm -p 0 -g 30.0 -M raw -F9 -s 48000 -f 405500000 | demod_stuff_here

For testing, you can actually set the path to the rtl_fm binary in the config file here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L395
With that changed to your program, you could then run auto_rx in 'full manual' mode using:
python auto_rx.py -m LMS6 -f 402.000

This bypasses the scanning and detection steps, and immediately starts a decoder for the supplied sonde type, on the supplied frequency

rs1729, 2019-09-01:
How about a lightweight (auto_)rx.py that only reads JSON-output and reports to sondehub or aprs (e.g. for testing ims100)? Then I could use other receivers.

darksidelemm, 2019-09-01:
That might be doable... let me have a think about how to implement that. One way might just be to have a 'dummy' sonde type which just listens on a UDP socket, so you can take your decoder and pipe it into netcat pointed at localhost. Would also need to override any callsign checks. 
Let me stew on it a bit and I'll try and implement something during the week or next weekend.

miweber67, 2019-09-02:
Sorry, I put this in the wrong thread before. Moving it...

OK, I have coded up a spyserver client that is a drop-in replacement for rtl_fm (for the purposes of auto_rx, anyway), although I didn't get a chance to test it thoroughly during tonight's sonde launches. Hopefully tomorrow I will have some more chances to dial it in. I am able to use the airspy dongle when running the spyserver on x86 Linux, but I wasn't able to get it to serve up my RTL-SDR dongle for some reason. Will have to keep looking in to that.

With respect to rtl_power equivalence... the spyserver will send "fft data" as well as IQ data; I am thinking that perhaps a separate client could pull the fft data for the entire band and perform the services currently provided by rtl_power. If that works, then we're really close to being able to cover all 6 MHz of the US sonde band with a single dongle and decode as many sondes as they see fit to launch at once, bound only by CPU.

I started with the 'spyserver' code from here:

https://github.com/racerxdl/gr-osmosdr/tree/master/lib/spyserver

I removed all of the gnuradio dependencies, and removed almost all of the boost dependencies (only circular_buffer is left) and made other adjustments to get things working. Not sure how to cite/credit the original; new to github so looking for guidance there.

darksidelemm, 2019-09-02:
Nice work!
It will definitely be interesting to see at what point we become CPU limited, especially if trying to process 6+ MHz of spectrum. Removing *all* boost dependencies would be nice if possible without a massive amount of effort...

For now, if the utilities could simply emulate the function of rtl_fm and rtl_power (with reasonable restrictions, i.e. perhaps no FM demod), that would certainly make things easier. I'll have to look at bringing in the latest dft_detect which can take IQ input, at the moment we need to feed FM-demodulated audio into it. 
Obviously it will make sense to change how auto_rx interfaces with these utilities at some point in the future, but for not a drop-in replacement for rtl_fm and rtl_power would be a good start.

rs1729, 2019-09-02:
You have to adjust to your own thresholds again...
Just added Meisei to dft_detect. The Meisei radiosondes could be false detected as iMet1RS(imet4), but that's a bit of a imet-problem, no unique header.

Many radiosonde types now to detect. Gets a bit messy.
For IQ-detection I still do FM-demodulation, so I don't have to deal with different modulation parameters, it is easier to detect that way right now. However for FM-demodulation it would be nice to do lowpass filtering before FM, but there are different possible signal-bandwidths, that's easier to filter in FFT for different bandwidth, but that's after FM at the moment. Maybe a wider 16 kHz filter might be ok for all, so have to test with weak signals, and frequency correction would also have less space.
Maybe the radiosonde types should be grouped in 2-3 bandwidth categories, processed in parallel... And then there are these afsk-radiosondes...

darksidelemm, 2019-09-02:
It's certainly not easy! I've already had to add in a LMS6 / RS92 switch in the config file for use with receivers listening on 1680 MHz. It's not possible to reliably detect them both with a single FM demod bandwidth due to the massive bandwidth of the 1680 MHz LMS6 sondes.

Do you have any long samples of the Meisei sondes that I could work with to optimize the thresholds?

vk3-hca, 2019-09-02:
How long do you want, 50megs ok?

darksidelemm, 2019-09-02:
Ideally I need an IQ recording, maybe 96 kHz bandwidth max (though wider is
OK, I can filter it), maybe 2 minutes long, centered on the sonde signal.
The sonde signal needs to be as high a SNR as possible, as that lets me add
more noise to test low SNR performance.

vk3-hca, 2019-09-02:
OK,
I had IQ as well, about 10 min files I think but needed to recover some disk space 
I mailed you a directory link with wav files and one untested IO.
More IQ data recording will have to wait till Wednesday week.

darksidelemm, 2019-09-02:
I have posted a link to the iMS files here: https://github.com/rs1729/RS/issues/14  (we're already off-topic enough as it is in this thread!)

miweber67, 2019-09-02:
Sorry for participating in the thread hijacking... is there another place we can discuss research and beta-approaches?

I am getting IQ output and generating fm-demod wav files that 'sound' reasonable, but decode performance is poor. IDK if I need to tweak the spyserver client itself, the fm demod, the sox resampling, or something else. Or it could be that the signal was just poor when I captured. Any advice?

Note: Because the airspy r2 has a native sampling rate of 10Msps and provides lower rates at decimations of 2^n where n is 0-11, the closest I can get to 48K is 78125. Half that is 39062 which does not seem to play well upsampling for the 48khz decode chain following.

rs1729, 2019-09-02:
Have an airspy mini, for auto_rx the gain-settings are important.
FM-demodulation (kind of time derivative) is sensitive to noise, so the IQ/IF-band needs filtering high frequencies outside the signal bandwidth.
With csdr you often see spikes in the fm-demod, even with bandpass/lowpass filter. After FM-demodulation lowpass filtering is also possible, but probably better before (FM is not linear). The signal should be centered in the filter window.

miweber67, 2019-09-09:
Just an update. Been working on the ss client a bit, and have discovered some behaviors that are sub-optimal but work-around-able. Still trying to find a good decode chain for the 78125 sps stream. Maybe I can post a sample file and someone can poke at it? I'm trying things like this:

cat 201909090806_401018000_g20_78125.s16 | sox -t raw -r 78125 -e s -b 16 -c 1 - -r 48000 -b 16 -t wav - | csdr convert_s16_f | csdr shift_addition_cc 0.0064 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr convert_f_s16 | play -v 0.5 -t s16 -b 16 -c 1 -r 48000 -

Hope to have the ss_client cleaned up to the point I can upload it in another few weeks.

darksidelemm, 2019-09-09:
Looks like you're trying to use sox to resample a complex data stream? sox won't know that it's complex data is is likely going to destroy the data in resampling.

I'd prefer to not require csdr as a dependency for auto_rx if at all possible... As nice as CSDR is (and I use it a lot for testing), the author hasn't really supported it in a while, and it's not properly cross-platform (i.e. getting it running under OSX, which is my primary platform) is a pain. 

For resampling signed-16-bit complex data you can try the tsrc utility which is within radiosonde_auto_rx/utils/ 

Your above command would become: 
`cat 201909090806_401018000_g20_78125.s16 | ./tsrc - - 0.6144 | csdr convert_s16_f | csdr shift_addition_cc 0.0064 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr convert_f_s16 | play -v 0.5 -t s16 -b 16 -c 1 -r 48000 -

miweber67, 2019-09-09:
tsrc did what I needed! Agreed about not adding csdr as a dependency; I was just trying to see if the data the ss_client was providing was decodable. I can now hear the sonde with the above command chain; now it's just a matter of tuning things.

Probably what I will do is just pull libsample into the ss_client and allow it to accept a request for an arbitrary sample rate, then pick one of the available rates and resample it internally. Progress!

darksidelemm, 2019-09-11:
A bit off-topic, but your (@miweber67 ) station managed to track two possible DFM-17 sondes almost to the ground : https://tracker.habhub.org/#!mt=satellite&mz=11&qm=3_days&mc=38.71678,-77.30913&q=RS_DFMxC-1808*

Recovering these and confirming that they are in-fact DFM-17 sondes would be very useful...

miweber67, 2019-09-15:
@darksidelemm , I check all sonde launch notifications to see if any of them are retrievable from both a landing site perspective and a "can I spend the time to drive all the way over there today" perspective. Sadly right now, many sondes are failing the first test, and most sondes are failing the second test. :-(  But, I am very interested in retrieving sondes, and I will give preference to the DFMxC ones that appear. It may just take a while. :-)

darksidelemm, 2019-09-15:
No probs - if you can switch to the testing branch and update that would be appreciated - it will now handle the 'LMS-X' sondes.

miweber67, 2019-09-15:
@darksidelemm , switched to 1.2.0-beta1 at commit edd45d0.

I spent some time today on the spyserver client, but went down a rabbit hole trying to get buffering to be more efficient and 8-bit samples working, so... not done yet. :-\

darksidelemm, 2019-09-15:
Hmm why 8-bit samples? Being able to output complex signed-16bit samples would probably be more useful, as that can go straight into the likes of fsk_demod, and i think @rs1729's demods too.

miweber67, 2019-09-16:
s16 works fine... I figured since the rtl tools were all 8-bit that that would be needed at some point. If not, I will stop worrying about it. :-)

Love that that SNR is now in the GUI! Now if I can just keep it from resetting to the left margin on mobile, I will be able to adjust antennas more easily. =D

darksidelemm, 2019-09-16:
Just note that the SNR is only there when using the fsk_demod-based demod chains!

I also only just noticed that Spyserver is closed source :-/

darksidelemm, 2019-09-16:
There is another SDR server program that should be getting released 'real soon now', and will actually be open source, and should have similar (if not more) functionality. Hopefully it won't be too hard to re-point at that.

rs1729, 2019-09-18:
> A bit off-topic, but your (@miweber67 ) station managed to track two possible DFM-17 sondes almost to the ground : [...]

Didn't remember where we discussed the possibility of DFM-09 w/ pressure sensor.
Here is one from Canada:
https://tracker.sondehub.org/?sondehub=1#!mt=osm&mz=10&qm=1_day&f=RS_DFMxC-18078803&q=RS_DFMxC*

[ 28] 2019-09-18 13:28:40.0   lat: 45.38621  lon: -75.82769  alt: 373.0  vH: 10.52  D: 306.4  vV: -14.11 
[ 29] 2019-09-18 13:28:41.0   lat: 45.38627  lon: -75.82780  alt: 358.8  vH: 10.60  D: 306.0  vV: -14.14  (ID-C:18078803) 
[ 30] 2019-09-18 13:28:42.0   lat: 45.38632  lon: -75.82791  alt: 344.6  vH: 10.52  D: 303.3  vV: -14.25 

[image]

More sensors, so the ID is in 0xC instead of 0xA.
Still don't know if the DFM-17 can also have 0xC or 0xD.

miweber67, 2019-09-20:
I posted my alpha version of the spyserver client here:

https://github.com/miweber67/spyserver_client

Still new at this so if I'm breaking conventions or etiquette please let me know.

darksidelemm, 2019-09-20:
Excellent work! Some information on how to use it in place of rtl_sdr and rtl_power would be good. I'm guessing it might need some intermediary shell scripts for compatibility?

miweber67, 2019-09-21:
Oops, yes I guess some documentation would be nice, eh? =D I updated the project readme.md with some usage instructions.

As far as compatibility, command line arguments are different, but the output of the fft mode should match the format of log_power.csv. One difference I need to fix is that the range of the report is not limited to a specific range; it's bounded by the center frequency and sample rate. That should be a relatively straightforward addition when I get a chance to do it.

The output of the iq mode is 16-bit signed ints instead of 8-bit unsigned, but otherwise is a stream.

What adjustments are most needed for auto_rx integration?

darksidelemm, 2019-09-21:
So first up, I don't think I really want to completely integrate it into auto_rx. I don't really like the idea of relying on a closed-source SDR server, with a protocol that could change and break compatibility at any time.

However, I think if it can be massaged into being a drop-in replacement for rtl_power and rtl_fm, I'd be happy for it to be an option for those that might want to try and cover the entire sonde band with (for example) one airspy/airspy-mini. I'd suggest that those users download your utilities, and change a few lines in the config file.

So a few points on how auto_rx currently uses RTLSDRs, and how I think we could abuse these methods to use your client.
- On startup, each SDR defined in the configuration file is checked for existence. This occurs here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/utils.py#L566  I think I'll add some lines so that if the device ID is something special, let's say 'TCPx' (where x is a number), then the function automatically returns True.
- When a Scanner or Decoder is started, the SDR is also re-tested. Same bypass as above applies.
- The Scanner will call rtl_power, but the filename for the rtl_power binary is set in the config file. So we can insert your replacement for rtl_power there. It will need to accept the same arguments, HOWEVER I think it should be able to safely ignore a lot of them. auto_rx does not check that the resultant spectral data produced by rtl_power matches the frequency range that was selected. So, if your rtl_power replacement just returns the appropriate format for the entire frequency range available from the SDR server, that would work. The rtl_power-emulated output needs to provide data on the entire receive bandwidth of the SDR, not just a small selection of it. 
- The Scanner uses rtl_fm, in FM Demodulation mode, with a custom receive bandwidth, to do the 'detection' step. I believe @rs1729 is working on adding IQ input support for detection, in which case you wouldn't need to implement FM demodulation. However it might be useful to have it available, because....
- Not all of the decoders can deal with IQ. All the decode chains that use fsk_demod use either 48 kHz or 96 kHz IQ input, however the decode chains for some of the more 'esoteric' sondes (iMet, LMS6-1680 as an example) only accept FM-demodulated input. Having some kind of FM demodulation option would be useful. Can the SpyServer give you FM audio?

Finally, the real benefit this kind of server-client approach will bring is the ability to have multiple clients running simultaneously, though they will of course be limited to operation within the available receive bandwidth of the SDR (i.e. the 3 or 6 MHz bandwidth of the Airspy Mini). So, being able to have multiple clients hook in and use the same SDR is pretty important! It would be good to know how many clients can be spun up on a single machine. For example, I would expect that you could have up to one scanner client running, and three demodulator clients running simultaneously, in a worst-case (i.e. parts of europe). 

It would be good to chat about this in a bit more detail - can you get on IRC at some point?

darksidelemm, 2019-09-21:
I've just merged a change into the testing branch that will cause rtlsdr_test to return True if the device id starts with TCP.

This means you should now be able to test out using your client as a replacement for rtl_fm/rtl_power. You'll need to set the device_idx field in your station.cfg to something like TCP01.

rs1729, 2019-09-22:
dft_detect and the *mod-decoders can decode IQ in baseband or IF for some time now.
If it is a centered IF band (i.e. decimated otherwise it will take a while), then it is
./dft-detect --iq `<iq_data.wav>`
For the decoders the options are --iq0 (FM-demodulation) or --iq2 (integration/DFT).
In baseband it is --IQ `<fq>` where `fq` is the relative frequency in -0.5 .. 0.5.
For the decoders, for the --IQ options filtering (option --lp) is recommended.

Just added IQ offset correction, some IQ-streams have some dc offset, that makes demodulation difficult, in particular for FM-demodulation.
And now frequency correction with option --dc is included (for testing).
For FM-demodulated audio it is easy to correct, but if you choose a wide IF-filter to leave space for offset, the additional noise is not good for the FM-audio.
For IQ-data the frequency correction can be done before filtering, so with --lp --dc no filtered iq-stream is needed. If you don't jump several kHz, the offset can be +-5kHz and more.
Though I'm not sure if a centered and filtered iq-stream would decode a little bit better, still don't trust offset estimation, how stable it is and if it can run in a deadlock, if something strange happens to the signal. And the lowpass filters are a bit wider to prevent cutting the signal if it is not centered.
If the receiver has no offset, the more problematic radiosondes are DFM-09 and iMS-100.

Anyway there are many parameters and possibilities for different situations. And probably I should do some documentation for the options.
Decimation and filtering needs more resources, probably there is room for optimization (that is still readable). But it works also for airspy-mini with 6MHz bandwidth here, on my computer still faster than real time...

For other radiosonde type, a simple FM-demodulation could be done Thoug iMet-1-RS and iMet-4 have different bandwidth (modulation index), different filters would be needed. (Updated the imet-decoder last week, it is simpler/fast now. On my test samples it gave at least as much frames as the older version.)

darksidelemm, 2019-09-22:
OK - I'll give dft_detect a go using IQ input and compare detection performance with the FM demodulator option. It would certainly make things a lot easier. 

I'm still fairly happy with using fsk_demod for the sonde types that it is happy with (RS41, DFM, M10, LMS6-400). It doesn't like the high modulation index of the LMS6-1680, and for some reason it's not happy with the iMS100 either (suspect modulation index too low?), but for the rest it works well. The latest beta of auto_rx makes use of the modem stats output from fsk_demod, and so it can display SNR, which is useful for comparing/optimising receive setups. The SNR estimator has been previously tested against generated signals, and appears to be fairly accurate.
At the moment I'm using the --bin option on the mod decoders that support that (RS41, DFM), and my bits_to_samples.py script for those that don't (emulating a FM-demodulated signal). Performance seems to be about the same as with an optimised FM demodulator, though with way better handling of frequency offsets (i.e. as long as the signal is contained within the receive passband, it will find and demodulate it with zero performance impact).

Obviously for iMet sondes (and really I'm only going to continue supporting the iMet-4 and the narrow-band variant of the iMet-1-RS in auto_rx, since the wider bandwidth one is obsolete), I'll continue using a FM demod. Not much that can be done with that one. Same goes for the LMS6-1680.

darksidelemm, 2019-09-22:
Ran a quick test over my existing sample set, to see how the detection performance compared to the existing FM Demodulator approach. Raw results are here: https://slexy.org/view/s20HvUzL78  (the dB value is Eb/N0. Still not convinced I have the M10 noise addition correct, but the rest are OK. Anyway, this is a A/B comparison.)

Detection performance is generally improved by up to 5 dB, with the exception of the iMet sondes, for which detection performance has dropped by ~2dB.  These results are with a perfectly centred signal. 

I did a test (just with the DFM sondes) where I intentionally offset the signal by 5 kHz. When using a FM demodulator, the detection point changed from 12dB to 14 dB Eb/N0. The IQ decoder did not detect anything, even at the highest Eb/N0 values.  It appears that detection performance falls off a cliff above about 1.5kHz offset. With the FM demodulator it falls away a bit more gracefully. Any thoughts on how we could correct for these offsets? (given that they will be unknown)

rs1729, 2019-09-22:
I want to use the IQ-decoding for different iq-stream source, rtl_sdr, rtl_fm, airspy, sdrsharp.
And then in for decoding several signals in one iq-stream, so for automatic detection the frequency correction can be helpful.

For dfm and meisei the frequency correction can be really helpful.
As far as I know, the ims100 has 100kHz frequency steps, but the sonde drifts like dfm and the frequency can be off for 5kHz (or more). Don't know which demodulator you use for ims100 right now. On sondehub the frequency is often reported 10kHz high, so if the radiosonde is transmitting somewhere around +5kHz off and the decoder is then set to +10kHz, it would be 5kHz off, don't know if this is to much?
The modulation index is higher than the dfm-09 index.
I think dfm has around h=2.2-2.4, ims100 around h=2.8.
The baud rate is very similar, so the higher modulation index also fits the higher bandwidth of 12-14kHz (dfm has 8-10kHz).
 For the decoders I usually set it a bit lower than what I calculated. The ims100 has not much pulse shaping, that is good for decoding (and also adds bandwidth).

In some recordings from @vk3-hca I noticed a IQ dc-offset, that can affect the demodulation, in particular when doing FM-demodulation. Don't know if this could have been the cause of problems (sometimes) when using csdr, don't remember a IQ-correction step and if the sample did have an offset. And maybe it affects your demodulator as well, though for iq-demodulation via integration (or kind of simple DFT, resp.) this is not so problematic as for doing FM-demod.

rs1729, 2019-09-22:
I have different filter settings for IQ-detection, maybe I have to adjust them for iMet.
Though there are two possibilities, imet1rs or imet4, forgot if I test both bandwidth.

For DFM with offset try:
./dfm09mod --ecc -vv --iq2 --lp --dc `<iq-data.wav>`
(if bandwidth is downconverted)
or
./dfm09mod --ecc -vv --IQ 0.0 --lp --dc `<iq-data.wav>`
(0.0 if it is approx at center).

Detection:
./dft_detect -v -c --iq --dc `<iq_data.wav>`
or
./dft_detect -v --IQ 0.0 --dc `<iq_data.wav>`

The --dc option is still optional, so correction is not always turned on.
Without --dc the filter cuts the signal.

miweber67, 2019-09-22:
@darksidelemm : Agreed, by "integrate" I meant "make available as a front-end option with just config file settings." :-) Happy to try getting on IRC at some point to chat! My timezone is UTC -0400; when's good for you? My motivation for this effort has been to cover the many sondes in flight here simultaneously with just one receiver.

The rtl_power format should be all set. I will adjust argument parsing to be transparent.

Spyserver does not provide any demodulated signals. I'd need to add some FM demod code to the client to do that. I'm sure that's possible, just not sure how big of a job it is. Will add to wishlist.

Thanks for the testing solution! Will continue to work on that. I have a fork of the project that I plan to work in; I hope to also provide some tweaks that I've made, like the power-mask sonde detection, available over there for folks to poke at if they're interested.

darksidelemm, 2019-09-22:
The --IQ 0.0 --dc option definitely improves detection performance at higher offsets. Up to 3 kHz or so was working OK, but at 5 kHz no detection.

I twiddled the IF filter bandwidth on line 825 of dft_detect.c to be 18 kHz, and detection performance at 5kHz offset is now on par with the 22 kHz FM demod approach (in fact, some sondes dont decode at all with the FM demod approach and those kind of shifts). Results with this IF bandwidth and 5 kHz offset are here: https://slexy.org/view/s2OsTaxpVO
The detection performance with this 18 kHz IF BW is *slightly* (1-2 dB) degraded when there is no shift, but it's still detecting at a lower Eb/N0 than we can decode at, so it's still well and truly fit-for-purpose.

Is it possible to make dft_detect report the estimated offset from DC in the output? That way we could do some adjustment of the demodulation frequency before starting the demodulator.

As it is, I think with the IF bandwidth set to 18 kHz as mentioned above, I could easily switch auto_rx to using IQ input to dft_detect with no performance impact over what we are doing now. Will need to check on the IQ DC offset issue when using rtl_fm IQ output. Is it the DC option that helps with this?

darksidelemm, 2019-09-22:
@miweber67 If this dft_detect IQ change works out, then you'd really only need the FM Demodulator to handle sondes that still require a FM demod step (at the moment iMet, LMS6-1680, and iMS100, though iMS 100 can probably be switched to a IQ-based demod). Still might be worth implementing, could definitely be useful. 

I'm UTC+0930, and I'm usually on for maybe an hour between 0730 and 0830, and then in the evenings between about 1800 and 2300 local. I'm 'Darkside' on #highaltitude on Freenode.

rs1729, 2019-09-22:
Yes, I was thinking, 4-5kHz offset is the limit.
With a wider filter it could detect more, but then it is easier to overlook a weak signal, or that an unwanted signal is in the window. There is always a trade off. I have also experimented with FFT frequency estimations, but it didn't look as reliable and responsive and accurate, and for higher offsets it would be like scanning after all. The scanner or initial frequency should be less then 5 kHz off, that is maybe a good requirement. When the signal is centered, it still can follow the drift more than 5kHz. Normally it will not jump in 5kHz steps.
Also the decoder does not jump all the way to the offset, it approaches the offset in 2-3 frames, if the offset is large, so don't count the initial frames. Maybe damping makes it more stable and less nervous.
For the decoders I have a wider acquisition filter and when locked, a more narrow filter.
For a very weak signal it would be good to have a very narrow filter, but then the signal has to be centered accurately. And if there is another signal very close, the filter would have to be steep, this would be more computationally expensive.
Also the different demodulation methods have pros and cons. Funny thing, the --iq2 does not need much filtering and is less sensitive to small frequency offsets, but then it drops. Though it is kind of cyclic for the two fm-frequencies, so for a certain frequency shift, it comes out inverted. Illustrates the DFT behind that.

For a perfectly centered signal, the additional offset correction cannot improve the result, so for a very weak signal, offset correction can make it worse. The choice of filter is a similar problem. A very strong signal does not need much filtering and it would be much easier to correct higher offsets.

So for now the parameter settings are optional. Could make filter settings optional as well, but there are so many options already. Maybe testing can give some better settings. (Testing IQ-data is a bit more work than FM-audio samples.)
For detection, the filter settings can probably be improved. The decoders have individual filter settings, for detection it is a bit different.
Didn't test detection for L-band. The rs92mod decoder has the --ngp option to set the modulation index 4 times higher and calculate the frequency in the config data right. For rs92-ngp and mk2a and also imet-1-rs I have to take another look at iq-detection. dft_detect is doing FM-demodulation to simplify things, it is easier processing for several possible radiosonde types. And for detection only, it is not so critical as for decoding. However for FM-demodulation it is not so good to have a wide IF filter.

> Is it possible to make dft_detect report the estimated offset from DC in the output? That way we could do some adjustment of the demodulation frequency before starting the demodulator.

Yes, I want to do that.
For FM-audio it is a problem, since you don't know how much FM-gain has been applied. But for iq-data the debug-data shows good results.

> Will need to check on the IQ DC offset issue when using rtl_fm IQ output. Is it the DC option that helps with this?

Maybe it is "-E dc" (dc blocking filter)? Would have to look in the source code.
The IQ-sampling can have dc-offsets in both I and Q. This dc shows up at 0 Hz (dc) in the FFT. You just remove the average to remove the dc spike, or the first (few) FFT bins, that's the same, the 0 bin is just the average. sdrsharp has the checkbox "Correct IQ", gqrx has "DC remove". Don't know if rtl_sdr has such an option.
Since I don't know from which source the iq-data is coming, I try to remove the IQ-dc when I read the samples.

If you downconvert the last meisei-2mhz sample from 18.09., you can see the offset in the samples clearly. Maybe it has to do with 8bit quantization, the samples decode fine if you subtract 127.5 instead of 128, but then 0V (silence) would not map to an integer value... The actual offset is -0.6 (from 128).
When you open the downconverted version in sdrsharp, it needs a couple of seconds to correct it.
I was wondering what the problem was and why it didn't detect the strong signal, until I have seen the IF in the time domain. So IQ correction is not only for cosmetic reasons.

sdrsharp shows the spectrum (baseband and IF) of IQ-recordings very nicely. Would like to have something like that also for linux and where I can browse through the waterfall. baudline does not work with bigger files and has some other issues, gqrx is not very stable, difficult to load files, and and it is difficult to see weak signals, similar in baudline. inspectrum looked promising. gnuradio is not simple enough for that task.
Any suggestions?

darksidelemm, 2019-09-22:
With regards to the DC blocking, I was wondering more how your code deals with it, but it sounds like it should be fine - I'll try and add an option to inject a DC offset in my test scripts (mainly auto_rx/test/test_demod.py) so we can see what it does.

Apart from GQRX, I don't really have any good options for looking at these kind of files. I have my own hacky python script ( https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/test/plot_file.py ) which I use to have a look at the spectrum of signals and figure out occupied bandwidths and the like, but it doesn't do waterfall plots.

Let me know if you need any IQ samples of the various sonde types, I have quite a few good quality sample sets now which I've been using for testing purposes. Eventually @TheSkorm plans to add automated testing for auto_rx, so I don't break things when I make changes :-)

rs1729, 2019-09-22:
I update the IQ-correction estimate once every second, for detection 1/32 of a second. I believe the offset is not changing much over time. Averaging over longer period gave better results, though for decoding I start with 1/32 and double every cycle until the period is 1 sec. So after 1/32 sec it starts working.

rs1729, 2019-09-23:
Tested dft_detect --iq for radiosondes that have *mod.c decoders (and the --iq/--IQ option; forgot rs92mod).

In FM-audio mode imet1rs and imet4 sound the same. In IQ mode only imet4 with narrow bandwidth is supported for now, and then IMET4 is the output. Only in IQ mode --dc shows a frequency offset estimate (for FM audio I don't know if the demodulator applied some "FM gain").
--dc corrects approx. +-3kHz, for a strong signal maybe more, and it depends on how wide the IF filter is chosen (lpIQ_bw). More filter bandwidth gives more space for frequency correction, but weak signals will not be detected then.
Maybe I will try a different approach with more individual filters.
+-3kHz is not so bad, more important is that the decoder can track the signal in a wider range once it has a lock. (c.f. trade offs mentioned above)
Did try offset estimation via FFT, it was a bit wobbly, and when the IF-band was already filtered, it was difficult to tell, what is noise inside the filter. So for some IF-recordings it didn't perform good.

IMET1AB is maybe not so important anymore, maybe the IMET1RS will be replaced by IMET4 as well, so I don't know if iq-detection of IMET1RS with 60kHz bandwidth has to be supported.

L-band has the same problem. And for LMS6-1680/Mk2a with its huge bandwidth, the decimation would have to be limited (same for IMET1RS) for baseband IQ.
Maybe an L-band option would be best, so one could focus on RS92NGP and LMS6-1680/Mk2a in L-band. RS92NGP needs 4 times the RS92SGP-settings.

REMARK:
dfm09mod has another sn-updated.
Did notice that DFMxC had some false (first/initial) IDs on sondehub. Hope this update gets the ID right from the beginning.

darksidelemm, 2019-09-25:
OK, these updates sound good! I'll try and test them over the next few days.

rs1729, 2019-09-28:
The L-band option is not so useful, since the bandwidths of RS92NGP and Mk2a/LMS6-1680 are so different, and for --IQ the decimation has to be limited for the latter.

The `--bw <kHz>` option (for setting IQ filter bandwidth) will be more useful for `--iq` and `--IQ <fq>`, this way the decimation limit can be increased, if the filter bandwidth is high.
Tested with LMS6-1680 (--bw 180), RS92-NGP (--bw 30 or --bw 32), and IMET-1-RS (--bw 80 (60kHz too small)).
For weak signals the right bandwidth is more important. Don't know if the --dc offset estimates are so accurate for e.g. LMS6-1680/Mk2a and IMET-1-RS.
Occasionally --dc was a bit short, when I observed the DFM drift with option -c.

The *mod radiosondes should work better, the --bw is not needed here. In principle the iq-detection should work, maybe not quite as good as pre-filtered FM, and I don't have so much appropriate test data. For baseband data the signal frequencies would have to be found first.

miweber67, 2019-09-30:
OK, we are almost there! I was able to get the spyserver client working for scan / FFT and also for dft_detect. I somehow did not get the right collection of arguments to lms6mod to get any actual decodes before the signal faded.

Screenshot of the FFT is up on my spyserver_client repo at the bottom of the readme.

For detect, I used this command line. I needed sox for the .wav header; I don't think it's needed otherwise?

```
../spyserver_client/ss_iq -g 18  -s 78125 -f 401386000 | ./tsrc - - 0.614400 |
 sox -t raw -r 48000 -e unsigned-integer -b 16 -c 2 - -r 48000 -b 16 -t wav - |
 ./dft_detect --IQ 0
SS_client_if: Setting IQ sample rate to 78125 stage 7
SS_client_if: Starting Streaming
sample_rate: 48000
bits       : 16
channels   : 2
IF: 48000
dec: 1
LMS6: 0.9044
```
What is the 0.9044 after the LMS6?

rs1729, 2019-09-30:
That's the detection score, how good the signal matches the LMS6 header. Everything above 0.9 is good, the thresholds for the different radiosonde types are around 0.7.

Just added the option `- <sr> <bs>` for reading raw IQ data. Sets channels=2, for IQ/iq-data only.
`<sr>` = sample rate
`<bs>` = bits per sample (8: u8, 16: s16, 32: f32)

Now you don't need sox for the wav-header, e.g. use
`./dft_detect --IQ 0.0 - 48000 16 iq_data.raw`
for 16 bit signed integer iq_data.

If data is already centered and downconverted, you can use the option `--iq` instead of `--IQ <fq>`.
For detection probably you don't need to resample the 78125. Though for decoding and on slower machines, a lower sample rate can be necessary.

@miweber67 
Is your data 16 bit signed or unsigned? Do you convert from 16bit unsigned to 16bit signed? Isn't unsigned only used for 8bit?

darksidelemm, 2019-09-30:
Ahh this is excellent, I was about to ask for that raw input option!  
What would also be useful is some way of outputting the peak result for *all* the correlators, over the entirity of an input sample (i.e. don't return as soon as one of the correlators is over a threshold, wait until the input file runs out, and *then* return. The reason for this is it helps with figuring out the minimum value a threshold can be set to, without causing false detections. I can then run through a whole set of noise and weak-SNR samples.

I implemented something like this a while back for [this note](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/test/notes/2019-03-01_dft_detect_optimization.md), but I can't remember how I did it now :-/

rs1729, 2019-09-30:
There is the `-c` option that shows all detections without stopping at the first result. Then you can choose the best result.

For the offset, the dimensionless value in -0.5 .. 0.5? (and both dimless and Hz-frequency as verbose data).
Example: (two signals at 403MHz)

a)
./dft_detect -c --dc --iq SDRSharp_20190703_063344Z_403000000Hz_IQ.wav
RS41: 0.9808 , +0.003595
M10: -0.8742 , +0.039400
RS41: 0.9800 , -0.008007
M10: 0.9012 , +0.048440
RS41: 0.9942 , -0.000360
M10: 0.9133 , +0.048085
RS41: 0.9974 , -0.002754
[...]

b)
[verbose]
./dft_detect -c -v --dc --iq SDRSharp_20190703_063344Z_403000000Hz_IQ.wav
sample: 31669
RS41: 0.9808 , +0.003595   [ fq-ofs: +0.003595 = +215.7Hz ]
sample: 44933
M10: -0.8742 , +0.039400   [ fq-ofs: +0.039400 = +2364.0Hz ]
sample: 91697
RS41: 0.9800 , -0.008007   [ fq-ofs: -0.008007 = -480.4Hz ]
sample: 104894
M10: 0.9012 , +0.048440   [ fq-ofs: +0.048440 = +2906.4Hz ]
sample: 151706
RS41: 0.9942 , -0.000360   [ fq-ofs: -0.000360 = -21.6Hz ]
sample: 164871
M10: 0.9133 , +0.048085   [ fq-ofs: +0.048085 = +2885.1Hz ]
sample: 211714
RS41: 0.9974 , -0.002754   [ fq-ofs: -0.002754 = -165.2Hz ]
[...]

Or rather
c)
./dft_detect -c --dc --iq SDRSharp_20190703_063344Z_403000000Hz_IQ.wav
RS41: 0.9808 , +215.7Hz
M10: -0.8742 , +2364.0Hz
RS41: 0.9800 , -480.4Hz
M10: 0.9012 , +2906.4Hz
RS41: 0.9942 , -21.6Hz
M10: 0.9133 , +2885.1Hz
RS41: 0.9974 , -165.2Hz
[...]

If both values are in the -v output, I don't know what to prefer. It is 
freq[Hz] = fq * sr_base (sr_base = input sample rate)

darksidelemm, 2019-09-30:
I think having it just in Hz is probably better - you already know the sample rate, so may as well show it as Hz. Then I can tweak the RX frequency before starting a decoder.

The -c option doesn't quite do what I'm after - I'm interested in knowing what the scores for *all* the sonde types, not just when its detected something.

[23 more replies]

closed 2024-11-16 by darksidelemm