projecthorus/radiosonde_auto_rx #155: iMet sondes do not show ascend / descend rate

zanco, 2019-04-04:
I noticed that the iMet sondes do not show ascend and / or descend rate in the web viewer of auto_rx. 

Did not check but happysat said the info had to be in the logged data because the linked chasemapper on his Pi did show the info.

rs1729, 2019-04-04:
Velocity is in the enhanced GPS data packet 0x05.
The imet-1-rs/imet-4 that I know only transmit GPS data packet 0x02, and (enhanced) PTU data packet 0x04.
https://github.com/rs1729/RS/blob/master/imet/imet1rs-binary.pdf
Have seen extra data packet 0x03, e.g. ozone sounding at Lindenberg.

zanco, 2019-04-04:
thanks for the info, that could mean that chasemapper calculates this ascend / descend speed just from altitude difference which I am now doing in my head.

Would like to have that added to the auto_rx if possible.

darksidelemm, 2019-04-04:
Not too difficult to add, I can do it in the same way as chasemapper, which just uses the time and altitude deltas. Not sure when i'll get around to it though.

darksidelemm, 2019-05-12:
Resolved in https://github.com/projecthorus/radiosonde_auto_rx/pull/172. This will be merged into Master soon, please test using the testing branch in the meantime.
All telemetry is now passed through a GenericTrack object (from chasemapper), which will calculate a running average of the ascent rate. It also produces horizontal velocity estimates, and headings, though we currently don't use those.

closed 2019-05-12 by darksidelemm