projecthorus/radiosonde_auto_rx #156: USB

DL2MF, 2019-04-05:
I'm using and working in depth with all current available radiosonde tools, but radiosonde_auto_rx is a nightmare. Only eye candy, but no reliability, this system is the only one I did setup during the last years, which always has trouble with loosing the SDR connection.

2019-04-05T21:08:16Z | ERROR | Scanner #NESDR001 - RTLSDR #NESDR001 non-functional - exiting.
-- | -- | --
2019-04-05T21:08:16Z | ERROR | RTLSDR - RTLSDR with serial #NESDR001 is not present!
2019-04-05T21:08:16Z | INFO | SDR #NESDR001 has been allocated to Scanner.
2019-04-05T21:08:14Z | ERROR | Scanner #NESDR001 - RTLSDR #NESDR001 non-functional - exiting.
2019-04-05T21:08:14Z | ERROR | RTLSDR - RTLSDR with serial #NESDR001 is not present!
2019-04-05T21:08:14Z | INFO | SDR #NESDR001 has been allocated to Scanner.
2019-04-05T21:08:12Z | ERROR | Scanner #NESDR001 - RTLSDR #NESDR001 non-functional - exiting.
2019-04-05T21:08:12Z | ERROR | RTLSDR - RTLSDR with serial #NESDR001 is not present!

Tested on several hardware platforms (Lenovo Notebook, MacMini running Linux, Raspberry Pi 3B+) all my setups are not 24/7 reliable. There are so many design errors in this tool, starting with the unnecessary fixed name assignment when using more then 1 SDR, why not learning from the other rugged tools? Why implementing an USB reset ig it's already available in RTL tools?

I'm located in germany and running one of the firsts gateways reporting to https://wetterson.de based on dxlAPRS, this gateway is working since more than 2 years with only rebooting when software update was required. I had also a 2nd and 3rd gateway reporting for several months to https://radiosondy.info based on a Raspberry Pi 3B+ and running 6 months in 24/7 operation.

All gateways are equipped with 4 USB-SDR for each of them, using genuine RTL-SDR.com and the NooElec NESDR smart, none did loose the connection to one of the SDR.

What is the problem with your solution?

darksidelemm, 2019-04-05:
We have plenty of stations that are running continuously with no problems. Maybe not with 4x RTLSDRs, but there's definitely one running with 3. Yes, the RTLSDRs have to be reset from time to time. 

The problems with the RTLSDR locking up *may* be related to the use of the various rtl-sdr utilities (rtl_fm, rtl_power) and not talking to the RTLSDR 'direct' in the code.
You mention 'USB reset is available in RTL tools' - where? Please link me to an example of what you mean by this.

You also need to remember that I work on this in my own time, and this is my first large-ish project. My aim was to write a radiosonde tool that I could actually read the code for and understand what's going on.

If you don't like it, either step up and help me fix the problems, or go use something else. There's nothing forcing you to use auto_rx.

closed 2019-04-05 by darksidelemm

TheSkorm, 2019-04-05:
@dl2mf can you provide a copy of rtl_test. Can you also ensure that no other sdr tools are running.

The fixed name setup is required so specifix configuration can be applied to each SDR for cases where different SDRs, or antennas are used, or when other SDRs are used on the same machine by other tools. It's also useful when you want to run two copies of autorx at the same for different antennas and have both monitoring the same  sonde. The configuration is very flexible but requires the serial number to be based to the respective downstream tools, rtl_fm ect...