projecthorus/radiosonde_auto_rx #173: Log to MySQL

dterweij, 2019-05-13:
Request
Possibility to log a sonde to MySQL database. 
So you can make your own page for statistics like, Closest distance, most far distance of a sonde, how many sondes in a period of time, lowest altitude, highest altitude.

Currently i cant scroll the sondes list to read info. If you have a lot sondes in the list it goes back to top when a refresh poll is made, so you cant read the information of al sondes.

So a logging to mysql database would be a nice option.

darksidelemm, 2019-05-13:
The scrolling issue will be fixed in the next update. 

As for logging to a database, I'd rather not add even more external interfaces at this point. It should be relatively trivial for you to write some code to read the log files which are stored to disk in at regular intervals. If you want the data live, then the [UDP interface](https://github.com/projecthorus/radiosonde_auto_rx/wiki/Configuration-Settings#payload-summary-output) is fairly easy to use as well.

closed 2019-05-13 by darksidelemm

dterweij, 2019-05-17:
Okay the UDP is usable, dont looks to me as a valid JSON object but ok, workable. Though I have to do all the math myself I see.
I could not find it yet, maybe someone can help with it.
Distance between My lat, long to Balloon lat, lon and the altitude. And I read something about earths rounding etc I expect it is some formula I did not learn at school :)
So if someone can give a formula/example... The parser will be PHP as I dont know much about Python, i try to hack the UDP example to send the JSON to a PHP file to parsewhay I want into a database.

darksidelemm, 2019-05-17:
There's a generic lat/lon position calculation function written in python here which probably does what you need: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/utils.py#L632

There's also a Javascript version of the same thing here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/static/js/utils.js#L31

This is what is used in the web interface to calculate the distance/bearing/elevation from the RX station to a sonde.

dterweij, 2019-05-26:
Thank you. I have now most of the data ready. Its now time to implement MySQL.
I dont need to pipe anymore, just all from python.

Current data output:

```
DT2STW_AUTO_RX P4030265 00:18:21 11054.21261 -55.5 403.900 MHz Distance: 110.6 km Elevation: 5.2 deg Bearing: 208.1 deg SSW
DT1STW_AUTO_RX P4030265 00:18:22 11059.92923 -55.5 403.900 MHz Distance: 110.6 km Elevation: 5.2 deg Bearing: 208.1 deg SSW
DT2STW_AUTO_RX P4030265 00:18:26 11080.08086 -55.7 403.900 MHz Distance: 110.7 km Elevation: 5.2 deg Bearing: 208.1 deg SSW
DT1STW_AUTO_RX P4030265 00:18:27 11084.41083 -55.7 403.900 MHz Distance: 110.7 km Elevation: 5.2 deg Bearing: 208.1 deg SSW
DT2STW_AUTO_RX P4030265 00:18:31 11104.28477 -55.8 403.900 MHz Distance: 110.8 km Elevation: 5.3 deg Bearing: 208.0 deg SSW
DT1STW_AUTO_RX P4030265 00:18:33 11114.66708 -55.8 403.900 MHz Distance: 110.8 km Elevation: 5.3 deg Bearing: 208.0 deg SSW
DT2STW_AUTO_RX P4030265 00:18:36 11127.6231 -55.9 403.900 MHz Distance: 110.9 km Elevation: 5.3 deg Bearing: 208.0 deg SSW
DT1STW_AUTO_RX P4030265 00:18:37 11131.7832 -56.0 403.900 MHz Distance: 110.9 km Elevation: 5.3 deg Bearing: 208.0 deg SSW
```

dterweij, 2019-05-26:
@darksidelemm Can you add 'frame' to the payload_summary?

dterweij, 2019-06-04:
@darksidelemm Mark, it goes well. A quick and dirty python script that needs rework by some better python coders.  Just left is a simple php webpage that shows some stats.  When done I would not mind if you add the script and webpage as example within your utils directory. So far this is what i have done,

Listen to UDP
Sonde detected? check if we have it already or not.
New? add data to table 1, add data to table 2
Not new? update data in table 1
Table 1 holds in the end of transmission the last data and table 2 holds the first data.

That is enough data to play with and make nice statistics like, max distance you ever reached, number of sondes, lowest altitude, highest altitude, and so on.

I hope you dont forget to add the 'frame' number to the payload :) I have implemented it already.

[https://github.com/dterweij/autorx2mysql](https://github.com/dterweij/autorx2mysql)

Regards,
Danny.

darksidelemm, 2019-06-04:
The frame field will be added to the JSON blob as of the next release.

You might want to have a look at the statistics available on radiosondy.info - You do need to create an account to access them though. It provides the same kind of data you are calculating.