projecthorus/radiosonde_auto_rx #179: make the UDP ouput valid JSON object

dterweij, 2019-05-18:
Current UDP output (JSON parse errors: 21)
```
{u'altitude': 19454.46723,
 u'bt': 10791,
 u'callsign': u'P0520296',
 u'comment': u'Radiosonde',
 u'freq': u'405.900 MHz',
 u'heading': 269.82559,
 u'humidity': -1,
 u'latitude': 53.02944,
 u'longitude': 9.09692,
 u'model': u'RS41',
 u'sats': 10,
 u'speed': 29.592792,
 u'temp': -273.0,
 u'time': u'06:51:29',
 u'type': u'PAYLOAD_SUMMARY'}
```

Valid JSON output:

```
{"stationcall":
{"altitude":"19454.46723",
"bt":"10791",
"callsign":"P0520296",
"comment":"Radiosonde",
"freq":"405.900 MHz",
"heading":"269.82559",
"humidity":"-1",
"latitude":"53.02944",
"longitude":"9.09692",
"model":"RS41",
"sats":"10",
"speed":"29.592792",
"temp":"-273.0",
"time":"06:51:29",
"type":"PAYLOAD_SUMMARY"}}
```

As you see I added a "stationcall"  because I have 2 receivers and there is no indification to see what receiver sended the object.

Maybe from station.cfg add more send types. Like UDP_TYPE=[Chasemapper,JSON,whatevertype]

So I can pipe like:
`danny@node3:~/radiosonde_auto_rx/auto_rx/utils$ python horus_udp_receiver_example.py | myparser.php`

darksidelemm, 2019-05-18:
Are you sure about the JSON being invalid? I've just ran some output from auto_Rx through JSONLint.com without issues. 
The output I used was captured using `nc -lu 55672`, which resulted in:
`{"comment": "Radiosonde", "callsign": "P4810726", "freq": "402.500 MHz", "speed": 3.72348, "sats": 4, "temp": -273.0, "altitude": 84.11076, "longitude": 138.69281, "humidity": -1, "time": "00:00:00", "latitude": -34.7207, "model": "RS41", "type": "PAYLOAD_SUMMARY", "heading": 151.11263}`

Adding the `stationcall` field (or perhaps just 'station' ?) makes sense.

darksidelemm, 2019-05-18:
Oh, and if you're trying to pipe the output from horus_udp_receiver_example.py into something else, then of course that will be invalid. That code receives the JSON blob, and then uses the Python prettyprint library to print it out in an easier-to-read format, which clearly isn't going to be JSON compliant (and it was never meant to be?).

darksidelemm, 2019-05-19:
I've added a 'station' field into the Payload Summary output in v1.1.1. This takes its value from the habitat_user_call configuration file parameter, until I can think of a better option.

closed 2019-05-19 by darksidelemm

dterweij, 2019-05-24:
Thanks for adding the station field. I have to dig in google to get me a good output then :)