projecthorus/radiosonde_auto_rx #184: Add some control over running processes to Web Interface

darksidelemm, 2019-05-23:
Add some additional controls to the web interface, to allow (as a start):

- Halting of a current decoder. (and optionally locking out the frequency)
- Manual start of a decoder on a particular frequency (via insertion of a scan result).

Endpoints for stopping/starting a decoder already exist here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/web.py#L143

However it is not safe to expose these to a web client without some form of authentication. Options include:
- Password defined in the config file, required to access the web interface.
- Could allow 'read only' access to the web interface without a password.

LukePrior, 2021-05-22:
I think this can be closed now unless there are plans to add more controls?

Moll1989, 2021-05-22:
I don't really intend to advocate for scope creep; but I think it could be useful to view/modify the whitelist/blacklist frequencies on this page.

For instance I blacklist 402.000MHz due to a local interference peak which always appears there.  I don't know if the source is permanent, but to be able to remove and add that to the black list from web interface would allow me to conveniently check it from time to time.

darksidelemm, 2021-05-22:
I'm quite hesitant to add the ability to change configuration file settings to the web interface..
Doing this means I need a way to write those changes back out the configuration file, which is not trivial with the current configuration file scheme.

In your case @Moll1989  you can safely add that file to the never_scan list, as the BOM don't have a license on 402 MHz, so we would not expect to see any sondes on that frequency.

closed 2024-11-16 by darksidelemm