projecthorus/radiosonde_auto_rx #188: Radiosondy and auto_rx different DFM serialnumbers

dterweij, 2019-06-09:
It seems that radiosondy and auto_rx uses different serial numbers.

I did a quick dirty replace in my UDP to MySQL logger webpage:

`
	// Seems radiosondy and auto_rx uses different serial numbers?
	// ToDo: DFM06 and IMET? is it always DFM09-18 to replace?
	if ($result[$i]['model'] == "DFM") {
		$sondenumber = str_replace("DFM09-18","DF9",$sondenumber);
	}
	echo '<td><a href="https://radiosondy.info/sonde.php?sondenumber='. $sondenumber . '" target="new">' . $result[$i]['callsign'] . '</a></td>';
`

darksidelemm, 2019-06-10:
Yes, we use more descriptive serial numbers for DFM sondes (DFM-06, DFM-09, and the newer DFM-17). I don't intend on changing the serial numbers reported via the UDP output field, as they currently match the manufacturer's model numbers. 

The serial numbers going to radiosondy.info (i.e. going to APRS-IS) are shortened as we have a tight limit (9) on the number of characters possible for an APRS object name. The code that performs this conversion is here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/aprs.py#L42

It is also quite possible that the serial numbers reported by other sonde RX software (mainly dxlAPRS) are different to that reported by auto_rx (see this issue: https://github.com/projecthorus/radiosonde_auto_rx/issues/118 ). 
I've tried to have a look at dxlAPRS's code, but have given up on figuring out what's going on. If someone wants to volunteer to figure out the algorithm used to calculate the serial numbers its using, i'd be happy to change auto_rx to be compatible.

closed 2019-06-10 by darksidelemm

dterweij, 2019-06-11:
Okay thank you for the response.

Found another cleuless difference:
radiosonde_auto_rx IMET-1CBA91B2
radiosondy IMETB127E
This one I can't figure out :(
 
And still i dont see myself and many other dutch on the radiosondy website. I think the APRS is broken somewhere?

darksidelemm, 2019-06-11:
The iMet sondes don't broadcast a serial number *at all* so we have to make one up based on information we do have access to. It's no surprise that other decoders might end up with different serials. See here for information: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Model-Specific-Notes#intermet-imet-1--imet-4

As for showing up on radiosondy.info, I recently (a few weeks ago) made some changes to auto_rx to make this work. Make sure you are on the latest release, and also make sure your APRS icon is set to the 'satellite dish'. See here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L195

For some reason you will only show up on radiosondy.info's station list if your icon is set to that. Just a limitation of radiosondy.info...

balrog-kun, 2019-08-30:
Hi, should this decoder still be compatible with radiosondy.info? (sorry for hijacking this issue # but I'm not sure if what I'm seeing is a new issue).  I'm detecting RS41 sondes using this code and I have the icon set to the satellite dish.  It reports successful uploads but the sondes neither show up on radiosondy.info nor are they forwarded to aprs.fi.

2019-08-30 12:29:30,166 INFO:APRS-IS - Uploaded to APRS-IS: ECKIY>APRARX,SONDEGATE,TCPIP,qAR,ECKIY:;P0740736 *122944h4022.69N/00337.78WO151/005/A=066508 Clb=6.0m/s t=-58.5C 403.700 MHz Type=RS41-SGP Radiosonde http://bit.ly/2Bj4Sfk !wR3!

darksidelemm, 2019-08-30:
Hi! Ye, it definitely should be working... In fact I have 2 stations running uploading to radiosondy at the moment (see R2221406). Not sure if something is broken on their servers...

balrog-kun, 2019-08-30:
Thanks for confirming! Maybe something's wrong with my login then.