projecthorus/radiosonde_auto_rx #198: Email notifications if sonde is xx km from specified position

inkiiin, 2019-07-23:
It would be useful if I can configure multiple positions with corresponding ranges in configfile and If a sonde is within the specified range autorx should send an email.

Best regards,
Inkiiin

darksidelemm, 2019-07-23:
Hi, while this is possible to implement, you might be able to get the same
functionality from other services, like radiosondy.info . Could you have a
look and see if radiosondy does what you are after?

Thanks,
Mark

inkiiin, 2019-07-23:
Oh, I don't know about that. I'll give it a try and let you know :) 

Thank you

inkiiin, 2019-07-23:
Checked ... while it is possible to configure notify's at a specified range for your own location, it is not possible to do this for multiple locations except you register multiple accounts to different locations.

In my case i monitoring 3 different "landing zones"  ;)

best regards,
Axel

darksidelemm, 2019-07-23:
I'd probably say that that's getting a bit specialised for auto_rx... The
configuration file is bad enough as it is, without adding this kind of more
advanced functionality.

You could probably write something that listens to the UDP output that
auto_rx will emit.

inkiiin, 2019-07-23:
Hi,
nice idea .. this semes to be a bit more modular, i'll try to implement this that way .

can be closed, thank you

closed 2019-07-23 by darksidelemm