projecthorus/radiosonde_auto_rx #203: web page map, respond on click

vk3-hca, 2019-09-05:
Small thing but its annoying rolling the mouse wheel from scan results up to current task on the webpage.
It stops halfway up and starts zooming the map but the map scale is fine and I don't want to change it, I just want to get to the top of the page. 
Maybe an "onclick" before jumping into map zoom.

If this is some dumb assed brower fault, apologies and forget I asked.

darksidelemm, 2019-09-06:
It's probably just me not setting leaflet options correctly. I've got limited time to work on this, so if you can figure out what settings are required and let me know that would be useful.

darksidelemm, 2019-09-09:
This is now fixed in the testing branch via this PR: https://github.com/projecthorus/radiosonde_auto_rx/pull/209

The testing branch will be merged into master in a few weeks, and I'll close this issue then.

nickoe, 2019-10-18:
@vk3-hca it looks like @darksidelemm 's fix was merged. I guess this can be closed.

snh, 2019-10-26:
This has landed in `master` and is available as part of v1.2.0 now. Please feel free to reopen if you come across any issues.

closed 2019-10-26 by snh