projecthorus/radiosonde_auto_rx #204: Signal level to UDP output

dterweij, 2019-09-06:
I would like a signal level of the sonde to the UDP packets output.

Possible?

darksidelemm, 2019-09-06:
Currently no, as I'm not reading in the modem statistics output from fsk_demod.

It's on the TODO list... just an issue of time.

darksidelemm, 2019-09-14:
The latest testing branch commit now has this included, *iff* you are using the experimental demodulator chains (works for RS41, RS92, DFM, M10 and LMS6-400).
It adds the following fields to the UDP output (note that these may not always exist - i.e. the first few packets may have them missing):
* 'snr' - Float - Estimated SNR, averaged over the packet length.
* 'fest' - List (float, float) - Modem tone estimates.
* 'ppm' - Float - Estimated modem symbol rate error, in parts-per-million.
* 'f_centre' - Float - estimated centre frequency of the radiosonde, signal, based on the SDR frequency and the modem tone estimates.

I'm yet to test how this change affects performance when running on a raspberry pi. Once i've done this, and merged the update into master, I'll close this issue.

closed 2019-10-26 by darksidelemm

dterweij, 2019-11-03:
Hello @darksidelemm ,

Running it for 2 days now. Looks great.
Just the reported Frequency is kind of messy.

Normal payload frequency: 405.300 Mhz
Experimental payload frequency: 405.302 Mhz (Depends on your SDR PPM also)
Experimental payload f_centre: 405301156.25

I would like to see Normal payload as 405.300 Mhz while f_centre shows the center freq in experimental mode. 

snr type is in : dB (0dB-40dB range)
fest type is in : Hz
f_centre type is in : Hz

I hope at least the snr can be put out of experimental mode when you feel it is not affecting performance. I do run it at (2 SDR receivers)  Ubuntu PC and a PI3. No problems so far.

darksidelemm, 2019-11-03:
I'd suggest just quantize the centre frequency to the nearest 10 kHz and use that - it's all I would be doing.

SNR is *only* available in experimental mode, as only those decode chains (i.e. using fsk_demod) can currently provide that information. I.e. i can't get the SNR information our of rtl_fm.
I have changed the defaults in station.cfg.example to have them enabled by default now.