projecthorus/radiosonde_auto_rx #205: Run Script Upon Detection

ZigiWalter, 2019-09-06:
Hi,

(I hope this is the right place for feature requests. If it's not - where would that be?)

First, thanks for a great program!
I've been using the email notification feature. Can you add an option to also call an arbitrary external script upon detecting a new sonde? This would enable further automation.

Thanks.

darksidelemm, 2019-09-06:
There are already a few 'generic' output interfaces that could be used to provide the functionality you are specifically after. In particular, it wouldn't be difficult to use the UDP output to detect when a new sonde shows up. 

There's example code which shows how to grab the UDP packets here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/horus_udp_receiver_example.py

I've got pretty limited time to work on this project at the moment, so I'd suggest having a go at implementing what you need yourself using that as an example.

ZigiWalter, 2019-09-07:
Thanks, I wasn't aware of the UDP interface; can give it a try.
For development and testing - is there a way to "fake" a sonde detection...?

darksidelemm, 2019-09-08:
At the moment, not really, but it is something I should probably add.

There are various log file playback utilities scattered throughout my repositories, for example this one from horus_utils that will play back a sonde log file: https://github.com/projecthorus/horus_utils/blob/master/testing/emulate_telemetry.py

You would need to run it with the --summary option which will cause it to emit UDP packets on port 55672.

Another fairly simple way would be to just use netcat and echo to emit packets, for example:

`echo {\"comment\": \"Log Playback\", \"altitude\": 500, \"longitude\": 138.0, \"callsign\": \"testing\", \"time\": \"00:11:22\", \"latitude\": -38.0, \"type\": \"PAYLOAD_SUMMARY\"} | nc -u localhost 55672`

ZigiWalter, 2019-09-14:
darksidelemm,
I gave it an initial try. In between your two answers, it seems that I'll be able to easily achieve what I'm after.
Thanks again!

closed 2019-09-14 by darksidelemm