projecthorus/radiosonde_auto_rx #207: DAO position error

dl9rdz, 2019-09-08:
I have the impression that the APRS position generation has a flaw regarding the DAO extension.  The DAO extensions is supposed to indicate additional precision digits that are **added** to the base position.

For the base position, however, autorx using **rounding** (to 0.01 minutes) to generate the base position, so for example N48°34.6770 becomes "4834.68N" in the aprs string.  After that, the suffix "70" is encoded in the DAO field.  aprs.fi then decodes this as location N48°34.6870.

If this interpretation of what is happening in the code is correct, I suggest replacing (in autorx/aprs.py)
`lat_min_str = ("%02.2f" % lat_minute).zfill(5)
`with 
`lat_min_str = ("%02.4f" % lat_minute)[:5].zfill(5)
`
and
`lon_min_str = ("%02.2f" % lon_minute).zfill(5)
`with
`lon_min_str = ("%02.4f" % lon_minute)[:5].zfill(5)
`
in order to use exactly the same rounding for the base position and the DAO suffix.

darksidelemm, 2019-09-08:
Hmm. I think you are right - good catch!
I'll fix this in the testing branch, which should be released in a few weeks.

darksidelemm, 2019-09-08:
Are you aware of any other 'verified' implementations of this, that I could use to perform some unit tests with?

darksidelemm, 2019-09-09:
I think you might have the order of the string slicing and the zfill the wrong way around.
What you proposed results in leading zeros not being included for numbers < 10.
i.e.:
```
>>> ("%02.4f" % 7.0001)[:5].zfill(5)
'7.000'
>>> ("%02.4f" % 17.0001)[:5].zfill(5)
'17.00'
```

Using this method retains the leading zeros:
`lat_min_str = ("%02.4f" % lat_minute).zfill(7)[:5]`

```
>>> ("%02.4f" % 7.0001).zfill(7)[:5]
'07.00'
>>> ("%02.4f" % 17.0001).zfill(7)[:5]
'17.00'
```

I've made this change in this PR to the testing branch: https://github.com/projecthorus/radiosonde_auto_rx/pull/208

This should be merged into master in a few weeks, and I'll close this issue when that happens.

closed 2019-10-26 by darksidelemm