projecthorus/radiosonde_auto_rx #214: RS41 not decode

fermars1974, 2019-09-13:
Hi, 
There must be a problem with my installation or something else because even do my Raspberry detect a radiosonde (RS41 in Spain) when start decoding it always gets a time out. 
Its a Raspberry 3B with a fresh installation (Raspbian Buster with desktop, downloaded yesterday and updated/pgraded) and just Radiosonde AutoRX installed following the instructions step by step. Everything went fine, no errors during installation.
It can detect 10 peaks on different frequencies everytime the software do a scan. I have a RS41 for testing and usually is detected on 403.700 (Radiosonde frequency) but sometimes it varies slightly. I also did the LTE_Cell_Scanner succesfully but may be CrystalCorrectionFactor values are not pretty good. Could this be the cause?

CID A      fc   foff RXPWR C nRB P  PR CrystalCorrectionFactor
436255    778M    24k -12.6 N 255 U UNK 1.0000308816545828705
440255    778M    57k -12.8 E 255 U UNK 1.0000732533705978256
488255    778M  50.5k   -13 E 255 U UNK 1.0000648786870893758
 14255    778M   -73k -13.1 E 255 U UNK 0.99990611901894210778

I also tried with different PPM values and Gain in status.cfg with no success.
To be sure my SDR is working properly I have decoded the Radiosonde in Windows using SDRSharp and RS41 Tracker or Sondemonitor,

What else could I try?
Thank you.

darksidelemm, 2019-09-13:
Those are *very* high PPM offset values, which suggests either you are using a non-TCXO RTLSDR, or LTE-Cell-Scanner is returning crap (which is possible). What kind of RTLSDR is it?

If you have a RS41 test sonde, what is the centre frequency as reported by SDR#? (with the receiver PPM set to 0) This will at least give you some better understanding of what the PPM value *should* be. 

You may also want to try turning on the [experimental decode chains](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L416), which are far more tolerant to receiver drift, but if your PPM really is that far off, I would suggest trying to figure out that issue first.

fermars1974, 2019-09-13:
I know PPM offset values are not good. My RTLSDR is this one:
https:// www.aliexpress.com /item/ 3033761055.html
(remove spaces)

It is supossed that is TCXO with a quite low PPM shift. Either way, I tried with some PPM values in status.cfg based on that LTE_Cell_Scanner values with no succeed. With PPM set to 0, frequency is usually detected in 403.730. I tried PPM values between -50 and +40 and sometimes I get the exact Radiosonde frequency but again, no success.

I have right now activated the experimental decode chains that are available at the bottom of the config file, setting it to True the RS41 decoding option but same result. 
There is a Radiosonde flying in this precise moment and it does not detect any... but I am decoding it "loud and clear" with my radio and RS41 Tracker on Windows.

This are the log messages whenever I start the program:

pi@raspberrypi:~/radiosonde_auto_rx/auto_rx $ python auto_rx.py
2019-09-13 12:38:17,000 INFO:Reading configuration file...
2019-09-13 12:38:17,009 ERROR:Config - Invalid or missing email settings. Disabling.
**2019-09-13 12:38:17,010 ERROR:Config - Missing new advanced decoder settings, using defaults.**
2019-09-13 12:38:19,885 INFO:Config - Tested SDR #0 OK
2019-09-13 12:38:19,892 INFO:Started Flask server on http://192.168.1.41:5000
 * Serving Flask app "autorx.web" (lazy loading)
2019-09-13 12:38:19,895 INFO:Telemetry Logger - Started Telemetry Logger Thread.
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
2019-09-13 12:38:19,906 INFO:APRS-IS - APRS Uploader Started.
2019-09-13 12:38:19,908 INFO:OziMux - Started OziMux / Payload Summary Exporter
2019-09-13 12:38:20,635 INFO:Version - Local Version: 1.1.2  Current Master Version: 1.1.3.3
2019-09-13 12:38:20,636 INFO:SDR #0 has been allocated to Scanner.
2019-09-13 12:38:21,577 INFO:Flask - New Web Client connected!
2019-09-13 12:38:23,520 INFO:Scanner #0 - Starting Scanner Thread
2019-09-13 12:38:23,525 INFO:Scanner #0 - Running frequency scan.
2019-09-13 12:38:26,767 INFO:Flask - New Web Client connected!
2019-09-13 12:38:44,351 INFO:Scanner #0 - Detected peaks on 10 frequencies (MHz): [403.21 400.41 402.26 405.11 403.01 404.16 403.41 401.36 405.06 403.2 ]

Do you see something wrong? Not sure if the error I bolded is important.

This is the rtl_test just in case there is something relevant.

rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 128 bytes

If you have some other suggestions, I am here with my ears opened :).
Thank you!

darksidelemm, 2019-09-13:
The aliexpress link doesn't work (even without the spaces). I suspect your SDR might not actually have a TCXO, if you are seeing that much variation.

Those bolded errors suggest that your station.cfg file does not have all the required parameters in it (and will result in the experimental decode chains not being enabled). Are you sure you have the most recent revision of auto_rx? Those errors were removed a few months ago...

Can you perhaps join #highaltitude on Freenode IRC? Debugging via github issue messages not the best approach...

fermars1974, 2019-09-13:
Well, you are right, I downgraded the soft version just to test it. Now is running the Radiosonde Auto-RX Status Version: 1.1.2. A friend of mine is using this one and works great. But anyway, with the latest release happen the same.

Please, try this link:
https://www.aliexpress.com/item/33033761055.html?spm=a2g0s.9042311.0.0.7fda4c4dmxMfUG

And this is a direct link to a capture, just in case:
https://i.postimg.cc/xCPpmZdr/Clipboard01.jpg

I though that the Raspberry is not powerful enough to power the RTLSDR, but I tried with several power supply and the error is the same. On the other hand, it works if I connect it to my computer USB directly to check it with SDRSharp.
I will try with another RTLSDR device, I will buy a new one and will test again. 

Thank you very much.

darksidelemm, 2019-09-13:
Perhaps open up the unit and check that it actually has a TCXO in it.. I've heard there are a lot of clones about the place that claim to have TCXOs and don't. 

Please have a look and see what frequency SDR# reports as the centre frequency for the sonde, with 0 PPM set in SDR#. You can then try and manually tell auto_rx to decode on a frequency (first setting the PPM value in station.cfg back to 0) by running `python auto_rx.py -m RS41 -f 401.xxx`. This will at least confirm that the decode chain is working at all. I would also try the experimental decoders, though if your SDR is really that far off-frequency even that might not work.

fermars1974, 2019-09-13:
Thank you, I will do that tests and report.

fermars1974, 2019-09-13:
I opened my device and..., mistery solved. My RTLSDR is a fake one :-(. The board is like the one is reported on this Russian website:
https://ur7hbq.blogspot.com/2018/12/rtl-sdr-v3-blog-fake.html
No need to understand Russian, is pretty clear.

Thanks again for your quick support and sorry to bother you with silly questions when the problem was just mine. One thing is clear, I have learnt a lot this time.
I going to buy a new one, hopefully not a fake one and I will try again. This time I am sure is going to work. If not, I will let you know but first I will have a look to the board...just in case.

Kind regards.

closed 2019-09-13 by darksidelemm