projecthorus/radiosonde_auto_rx #218: Log to KML

fermars1974, 2019-09-18:
Hello,

The more I dig into this piece of software, the more I like it!
I am trying to convert the LOG file with all the Radiosonde data into KML to draw on Google Earth the trajectory based on coordinates and altitude.
I have seen a python program on /utils/log_to_kml.py directory but I am not sure if this file is for that purpose. Otherwise, is there an easy way to do it? :) 

Thank you!!

darksidelemm, 2019-09-18:
The log_to_kml.py script is intended to convert saved log data (i.e. the files saved into the log directory) into kml files, yes. 
You can run it with something like (and it's been a while since I used it):
` python utils/log_to_kml.py -i log/path_to_your_log_file.log -o output.kml `

You also need to install a few libraries first using `sudo pip install fastkml shapely`

If you're after something to view the sonde data in google earth *live* then this script I wrote a very long time ago *may* work: https://github.com/projecthorus/horus_utils/tree/master/apps/kml_chase

Note that using this will require you to set your [payload_summary_port](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L242) config file option to 55672, as this is the fixed UDP port used by all my old horus_utils code. You would then use the --summary option on kml_server.py

Hope this helps!

I don't know if that script still works, as after I wrote that I moved on to developing [chasemapper](https://github.com/projecthorus/chasemapper).

fermars1974, 2019-09-19:
Sorry Mark for may late reply. 
Amazing, it works perfect! It convert the log into kml file almost instantly.
The only thing I did is install the "libgeos" library, apt-get install libgeos++, otherwise you get an error.

After that, just sudo pip install fastkml shapely

Then, you can use the .py script right away.
Thank you!

darksidelemm, 2019-09-19:
Glad to hear you got it working!

closed 2019-09-19 by darksidelemm