projecthorus/radiosonde_auto_rx #232: Notification Email issues

fermars1974, 2019-10-10:
Hello Mark, 

I have upgraded my raspberry to Buster. In Stretch I were using "ssmtp" and "mailutils" to send emails from console, but "ssmtp" is no longer maintained and it does not work on Buster so I have installed "msmtp" instead. After that, I noticed that emails sent from radiosonde_auto_rx when a new radiosonde is detected and decoded do not work anymore. I do not really know if it is something I miss on my Raspberry configuration or something have to be done on the program itself. I have looking into /var/log/mail.log and mail.err but no errors are shown there as far as I see.
Is there any other file to search and find clues? or may be you already detected this issue as well.

Thank you for your suggestions and support.
Regards.

darksidelemm, 2019-10-10:
The email notification is known to work OK with regular and SSL mail servers, so unfortunately I suspect this is your specific configuration. 
I haven't used the utilities you mention, so I can't really help...

fermars1974, 2019-10-10:
Ok, thank you. Is there any way to trigger the email send feature in order to debug the issue a bit quicker? Is it reflected on any log? If I find out the solution I will post it here for if anyone find it useful.

Regards.

darksidelemm, 2019-10-10:
Running the command: `python -m autorx.email_notification` from within the radiosonde_auto_rx/auto_rx directory should send a test email.

fermars1974, 2019-10-11:
This issue can be closed. 
I tested with a simple python script the functionality of the "smtplib" library and it worked. After that and many other tests, I reconfigured the station.cfg and change the original destination address to another different one and It worked too, so it was clear, the problem was on my first "to" address: a misconfiguration in a filter I created few days ago, I deleted and everything is working fine again. Your script is sending emails again fast and easy.
Thank you and sorry for the inconvenience.

closed 2019-10-11 by darksidelemm