projecthorus/radiosonde_auto_rx #239: RuntimeError: dictionary changed size during iteration

nickoe, 2019-10-18:
I am new to radiosonede_auto_rx. I just my first radiosonde packets this morging, but auto_rx.py crashed.

I just got this runtime error:
```
2019-10-18 06:46:31,639 INFO:Habitat - Uploaded sentence to Habitat successfully: $$RS_DFM09-18021588,214,04:46:27,57.06846,10.07972,147,9.1,10.3,-1.0,DFM DFM09-18021588 401.840 MHz*4C6A
2019-10-18 06:47:00,159 INFO:APRS-IS - Uploaded to APRS-IS: OZ3RF>APRARX,SONDEGATE,TCPIP,qAR,OZ3RF:;DF92FCD4 *044634h5704.13N/01004.75EO318/012/A=000259 Clb=-9.3m/s t=9.2C 401.840 MHz Type=DFM09 Radiosonde http://bit.ly/2Bj4Sfk !w:>!

2019-10-18 06:47:01,467 INFO:Habitat - Uploaded sentence to Habitat successfully: $$RS_DFM09-18021588,221,04:46:34,57.06888,10.07922,79,6.5,9.2,-1.0,DFM09 DFM09-18021588 401.840 MHz*D23C
2019-10-18 06:47:05,486 INFO:Telemetry Logger - Closed log file for DFM09-18021588
2019-10-18 06:48:37,560 INFO:Flask - New Web Client connected!
2019-10-18 06:49:35,471 ERROR:Decoder #0 DFM 401.840 - RX Timed out.
2019-10-18 06:49:36,473 INFO:Decoder #0 DFM 401.840 - Closed decoder subprocess.
Traceback (most recent call last):
  File "auto_rx.py", line 741, in <module>
    main()
  File "auto_rx.py", line 721, in main
    clean_task_list()
  File "auto_rx.py", line 328, in clean_task_list
    for _key in autorx.task_list.keys():
RuntimeError: dictionary changed size during iteration
Main Loop Error - dictionary changed size during iteration
2019-10-18 06:49:37,463 INFO:Web - Flask Server Shutdown.
2019-10-18 06:49:37,463 INFO:Starting shutdown of all threads.
2019-10-18 06:49:37,666 INFO:Telemetry Logger - Stopped Telemetry Logger Thread.
```

I attached my configuration file.
[station.txt](https://github.com/projecthorus/radiosonde_auto_rx/files/3742560/station.txt)

It did receive 3125 packets from a sonde, that sounds quite good to me.

snh, 2019-10-18:
Hey @nickoe :wave: Welcome to `radiosonde_auto_rx`!

This particular error has been fixed by https://github.com/projecthorus/radiosonde_auto_rx/pull/211, which is currently available in the `testing` branch of `radiosonde_auto_rx`. I believe @darksidelemm hopes to merge `testing` in to `master` soon, but in the mean time you are welcome to use the `testing` branch to take advantage of this fix.

Alternatively, as this error only occurs in Python 3, using Python 2 is another workaround.

nickoe, 2019-10-18:
@snh ok, thank you. That saves me some time. I guess I will try to switch over to testing when I get home. I will close this issue for now.

closed 2019-10-18 by nickoe

snh, 2019-10-26:
@nickoe FYI these fixes are available in `master` now and have been released as part of 1.2.0 if you want to head back to a stable/release branch.

nickoe, 2019-10-26:
@snh, thank you. I will switch back to the master branch. It seems to work really well.