projecthorus/radiosonde_auto_rx #241: Sonde(?) Identification - Help Needed

ZigiWalter, 2019-10-23:
Hi,

(Not sure that this is the appropriate place for this. If not - where is...?)

Need some help with understanding this signal I'm picking up lately. It has the properties of a Sonde (comes and goes, frequency is 403MHz, audio resembles known sondes), but I couldn't get it decoded.
Auto_rx identifies it as an IMet, allocates a decoder, then give up after some time without decoding. I've also tried all decoders in SondeMonitor with no success.

Here are links to two samples I have recorded:
https://media-upload.net/file/257a62777df0e6bb4eb0c0640b186d46
https://media-upload.net/file/0f695c7df8357b18076d6b8a5a5c8a48

Where I live, auto_rx easily decodes IMet, RS41, and DFM sondes we have flying around, but this one remains a mystery.

Thanks.

darksidelemm, 2019-10-24:
A few questions:
- Does it appear around sonde launch times? (i.e. 1115Z, 2315Z - and then disappear a few hours later)
- Where are you receiving from?

It looks like pulses of 2400 baud FSK, but the packet sizes appear very very short, probably too short for a radiosonde signal. There's a lot of other stuff that lives in that band, so it might not be a radiosonde.

ZigiWalter, 2019-10-24:
Thanks for the reply.
1. No - times do seem to be sync'ed with WMO launches. However, many of the radiosondes we pick up here are launched by other parties, rather than the local meteorological service, so it's not a clear-cut indication.
2. Assuming that by "where" you refer to my location, then it's callsign ZWxx_AUTO_RX there: https://tracker.sondehub.org/?sondehub=1#!mt=osm&mz=8&qm=6_hours&mc=31.89095,35.16127&f=none&q=RS_*;*chase

bazjo, 2019-10-24:
For me this sounds like a DFM-09 where the GPS has failed. Sometimes in this cases the sonde only sends brief spikes of modulation. I received a similar signal last week from the artillery site baumholder from the german army. I have a recording of this as well, but only on my computer at home, I'll try to upload it this evening as well

bazjo, 2019-10-27:
I hve a bit more investigation. My recording is under https://drive.google.com/file/d/1FfqIzqi_QAElnJfIWCPX0PLQEVcSHAQo/view?usp=sharing
However, i dont think its is a DFM-09 anymore as the structure is quite a bit different. Next step would be look at the transmission in more detail and look what structure it has

ZigiWalter, 2019-10-27:
@bazjo - thanks for trying! Indeed, I don't think it's a DFM too.

I'm currently puzzled by two questions:
1. What is that signal? I seem to pick it up in the same hour each morning - will continue tracking.
2. Why causes the false IMET identification? The only solution I have for now it to blacklist that frequency (403.0MHZ), but unfortunately it's a very popular frequency for RS41 sondes, so I'm looking for a better solution.

Any help with any of the above would be greatly appreciated :-)

ZigiWalter, 2019-11-01:
Still a no go. It still pops every now and then, but not decoded. I'll block it for now.

While we're at it, any chance someone has a clue about the following recording? It sounds different than the one above (picked up on 402.0 MHz), but is also not identified by AutoRX/SondeMonitor.

https://media-upload.net/fUp/b9686d5ca33cc8cd5c0d7ee83764305b.wmv
Any chance it is indeed a radiosonde?

Thanks!

darksidelemm, 2019-11-02:
From your location and this map: https://user-images.githubusercontent.com/10279523/60766636-ba2f6c80-a0ac-11e9-85d8-14a85906fa9f.jpg

The sondes that are launched in your area are:
- RS41 (Israel)
- DFM09 (Jordan)
- IMS-100 (Turkey)
There also appears to be M10 launches from Egypt, and from Iran. There are other countries nearby that may be launching sondes of types we don't know about, but they are not reporting to the WMO it seems. 

It's possible that it is a radiosonde, but if it's not being detected correctly, then it's definitely not something that we know about!

ZigiWalter, 2019-11-02:
Indeed, we pick up all of the above types, plus IMETs (Israel) and IMS-100 (Cyprus) - it's a real sonde zoo here. All of those are successfully decoded by AutoRX.

Well, until there's a breakthrough, I'll consider these two on 402/403MHz to be our Loch Ness monsters... I guess the issue can be closed.

Thanks.

closed 2019-11-02 by darksidelemm

ZigiWalter, 2019-12-04:
Just a minor update regarding the 403.0 MHz signal discussed in the first post above - it's definitely NOT a sonde. In the last week or so, it is received almost constantly, very often identified as an IMet and is allocated a decoder.
Since 403.0 MHz is a popular frequency for real sondes, I didn't want to black-list it altogether. Instead, I ended up implementing a "brown-list": if a decoder is allocated but no valid packet is decoded, the frequency is added to the built-in temporary block list. Not an ideal solution, but better than any alternative I could think of.

darksidelemm, 2019-12-04:
I thought about that exact same solution a while ago, but never implemented it as I was concerned it might block sondes that are just above the threshold for detection to work, but not quite high enough for a valid decode.

ZigiWalter, 2019-12-04:
I was concerned with the same, and have actually observed it happening once. However, as I've set the block time to be relatively short (15m), it's still better than the alternatives...