projecthorus/radiosonde_auto_rx #244: Add RSSI indication to web interface

mclane, 2019-10-27:
It would be nice to have a kind of RSSI indication in the upper part of the web interface. Otherwise: great job! I like it!

snh, 2019-10-27:
Hey @mclane 👋 

The latest release (1.2.0) displays the SNR in the web interface when using the [experimental decoders](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/test/notes/2019-04-26_fsk_demod.md). These are also logged (if enabled), and included in the UDP broadcast based payload summaries (https://github.com/projecthorus/radiosonde_auto_rx/issues/204#issuecomment-531461695).

If not already enabled, you can enable these by [updating your `station.cfg` configuration file](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L400-L413).

This can be used to help align your antenna, and to compare the performance of different antenna, filter, and lna combinations.

mclane, 2019-10-27:
Hi @snh ,
thanks for the hit; that was what I was looking for! So the issue can be closed.

closed 2019-10-27 by mclane