projecthorus/radiosonde_auto_rx #245: Auto-Generated IMET Call Signs (release 1.2.0)

ZigiWalter, 2019-10-27:
Hi,

It seems that with the latest release (1.2.0), auto_rx is locking on the actual frequency, even if "out-of-tune". This obviously improves decoding and range - a big thanks for the authors!

There is a minor issue though with IMET sondes: apparently, since the signal peak is not necessarily exactly the same for two receiving stations (due to SDR variations, transmitter drift, etc.), the same IMET gets uploaded twice to SonderHub using different call signs.

In the few cases observed since morning, the differences in frequency were only in the least significant digit (e.g., 404.998MHz vs. 404.997MHz). If so, a simple mitigation would be to round the frequency when generating the IMET call sign, but maybe there are better solutions.

Thanks.

darksidelemm, 2019-10-27:
Ahh now that's an unintended consequence I didn't pick up when testing. Good catch!

Yes, I agree that the frequency should be rounded. In the case of the iMet sondes this can be to the nearest 100 kHz, as the iMets have very coarse frequency steps.

I've just pushed a change to both testing and master branches that should fix this (v1.2.1). If you have contact with any of the other imet sonde receiving stations in your area, you might want to poke them to update!

closed 2019-10-27 by darksidelemm

ZigiWalter, 2019-10-27:
Thanks.
As for "_If you have contact with any of the other imet sonde receiving stations in your area, you might want to poke them to update_!", I have, and already did :-)

Thanks again!

Hoshen-4Z7HKA, 2019-10-28:
Well, we have both updated to 1.2.1, and the problem seems to be fixed. 
Also need to mention that the decoder is now performing much better...

Thanks!

ZigiWalter, 2019-11-02:
I've bumped into a semi-related issue, probably with lower priority:

It seems that peak frequencies check for a perfect match with the ones in the blacklist.
As a result, I occasionally see that when a sonde frequency gets blocked (e.g., an encrypted sonde at 405.0MHz), the scan process ends up decoding a near by frequency (e.g., 404.999MHz), which is not blocked. There, decoding does not succeed (as it's out-of-tune), so a decoder gets allocated and released in an almost endless loop.

Would it be a good idea to have some tolerance allowed when comparing the peak frequency with the blacklist? Or are there any negative side-effects I have missed?

Thanks.

darksidelemm, 2019-11-02:
Good catch again - another unforseen side-effect of the change to the updates. I think a tolerance when checking against the blacklist is probably the correct approach. I'll implement that and get an update out this weekend.

ZigiWalter, 2019-11-02:
Great - thanks!

If you're around that piece of code "anyway", please also consider formatting the display of the peak frequencies list using "%.3f" (I think that currently the values get rounded when displayed, which is somewhat misleading).

Having written that - I must write that the software is absolutely great, and the particular change in the release (while triggering some minor requests) is greatly appreciated - we can really see the difference. Thanks again!

darksidelemm, 2019-11-02:
Looking at the codebase, the peak list is just cast to a string without any special formatting. This means it will print as many decimal places as are required, i.e.:
```
>>> a = [401.500, 401.501]
>>> str(a)
'[401.5, 401.501]'
```
So if there's 1 kHz offsets, then it should show it. 

Could you please switch to the testing branch (git checkout testing) then run a git pull and restart. This should get you onto version 1.2.3-beta1, which should hopefully fix the issues with temporary blocks not being respected.
The new behaviour now will remove a detected peak if it is within +/- quantization/2.0
Example (with quantization value set to 10000 Hz):
- Temporary block added for 405001000 Hz ('actual' frequency of an encrypted sonde)
- Peak detected on 405000000 Hz - this will be blocked
- Peak detected on 405001000 Hz - this will be blocked
- Peak detected on 404999000 Hz - this will be blocked
- Peak detected on 405007000 Hz - this will not be blocked

Try it and see how you go!

ZigiWalter, 2019-11-02:
Thanks - testing now.
As for the formatting issue - either I was imagining things (sorry...), or it's a problem with the minor changes I did in the code. Will double check.

Thanks.

ZigiWalter, 2019-11-03:
As for testing - it ran over night. So far - so good. Thanks for the very quick fix!

darksidelemm, 2019-11-03:
OK - let's give it a few days to be sure. 
Keep a lookout any encrypted sondes, and make sure auto_rx behaves as expected with them (i.e. blocks them, and doesn't allow peaks to be detected within +/- 5 kHz of the blocked frequency).

ZigiWalter, 2019-11-03:
to be honest, I've never seen an encrypted sonde. I've bumped into the above while doing a quick&dirty hack to allow "location-based filtering" (IIRC, that feature was quested before, but the decision was not to implement).

It occurred to me that the synchronized nature of WMO launches means that remote, interesting/exotic sondes are consistently shadowed by the local launches (higher power, longer reception period). Since I have a single well-placed antenna (roof top), the existing multiple SDR support was not good enough.

Basically, I've duplicated the logic for detecting an encrypted sonde: (after some grace period of successful detections) the frequency is added to the block list and the decoder is closed. I simply didn't want to "pollute" this thread with this issue, so used "encrypted" as an example.

darksidelemm, 2019-11-03:
OK, so is the idea to block reception of the 'common' local sondes, to try and receive the ones from further away? I'm guessing your local sondes don't have consistent transmit frequencies, so you can't block them in any other way?

ZigiWalter, 2019-11-03:
Exactly.
I had to block according to some geographical bounding box, rather than by using the frequency blacklist.