projecthorus/radiosonde_auto_rx #248: Add coordinates & height of the last position mail notification

JRus-design, 2019-11-01:
Hi, 
Good day from Spain. First of all,  congratulation to Mark Jessop and all GitHub contributors to the project for this piece of software!
I'm a sonde hunter and member of the Spanish Sonde Hunter Group named "Sondas Meteorológicas". 
I have an implementation that can be useful for the sonde chasing: the notification by email of the coordinates & height of the last position of the received sondes, with a link to google maps to make more confortable the search.
What do you think?
Thanks to all & good Sonde Chase!
Javier

sp3rat, 2019-11-01:
Hi,it is very goog idea  :)Jurek sp3ratWysłano z telefonu Samsung
-------- Oryginalna wiadomość --------Od: JRus-design <notifications@github.com> Data: 01.11.2019  18:29  (GMT+01:00) Do: projecthorus/radiosonde_auto_rx <radiosonde_auto_rx@noreply.github.com> DW: Subscribed <subscribed@noreply.github.com> Temat: [projecthorus/radiosonde_auto_rx] Add coordinates & height of the last position mail notification (#248) Hi,

Good day from Spain. First of all,  congratulation to Mark Jessop and all GitHub contributors to the project for this piece of software!

I'm a sonde hunter and member of the Spanish Sonde Hunter Group named "Sondas Meteorológicas".

I have an implementation that can be useful for the sonde chasing: the notification by email of the coordinates & height of the last position of the received sondes, with a link to google maps to make more confortable the search.

What do you think?

Thanks to all & good Sonde Chase!

Javier

[

{

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/projecthorus/radiosonde_auto_rx/issues/248?email_source=notifications\u0026email_token=ACXSXFQIUHE37BXHNQKCFDLQRRRQ7A5CNFSM4JH5ZLFKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HWFKLYQ",

"url": "https://github.com/projecthorus/radiosonde_auto_rx/issues/248?email_source=notifications\u0026email_token=ACXSXFQIUHE37BXHNQKCFDLQRRRQ7A5CNFSM4JH5ZLFKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HWFKLYQ",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

JRus-design, 2019-11-01:
Define the end of the sounding it's a bit difficult, because you can lost the signal at 13000 Mtrs and receive it again at 4000 Mtrs after 1 hour... I think is necessary to detect the descending and a threshold selectable in the station.cfg, per example 2000 Mtrs and 10 min without reception.

JRus-design, 2019-11-01:
Only I have the idea...I have no programming level to implement it.. :(

darksidelemm, 2019-11-01:
Yes, what you are after is not exactly simple, because as you suggest, you may lose the sonde signal and re-acquire it later.
Also, are you really going to want to receive e-mails about sondes that have landed many hundreds of km away? 

I'm not sure when I'll have time to work on this. Maybe someone else can figure out how to modify the email notification code to do it and submit a pull request...

JRus-design, 2019-11-01:
Thanks Mark for the answer,

The mail can be triggered when we have:
*Negative vertical speed
*Sonde height below x (to be defined in station.cfg)
*Sonde distance from station less than x (to be defined in station.cfg)
*No signal received since x minutes (to be defined in station.cfg)

The mail then can contain sonde type, ID, coordinates & height of last
position with an hyperlink to Google Maps.

bazjo, 2019-11-01:
german site wettterson.de sends out notifcations when the sonde first is below 6000m, but provides a dynamic gmaps link through their site so that it always shows the latest position, it also has a link to a tawhiri-prediction based on the latest position. It provides email notification as well as a telegram bot. 

maybe some things to consider as well

darksidelemm, 2019-11-02:
Yes... rather than adding massive feature creep in this project, it might be good to see what existing services are out there. I'm not sure if auto_rx interfaces correctly with wetterson.de, but it definitely works with radiosondy.info, which I believe has similar functionality.

bazjo, 2019-11-02:
radiosondy.info notfications are inferior compared to wetterson.de ones, for the features described above. wetterson.de uses a modifed dxlAPRS for it's receive stations and does not want people to tranmit data withou prior approval from them, so just sending auto_rx data to those is not the best solution at the moment.

but i totally agree that developing everything three times is not a sustainable solution. one reason more to get all the developers together and create something awesome

darksidelemm, 2019-11-02:
I'd certainly be more than happy to make sure my APRS data conforms to what wetterson.de needs. I just need information what I should be conforming to!

bazjo, 2019-11-02:
as wetterson.de is not open for anyone to transmit data to in the moment, i don't know whether this is what jan and matze envision for their system, although i would really like it. although i can see why they are doing it this way, given the chaos with dfm etc.

darksidelemm, 2020-03-20:
Closing this issue for now. The DFM issues should be resolved soon, and hopefully the wetterson.de devs will let us into their walled garden...

closed 2020-03-20 by darksidelemm