projecthorus/radiosonde_auto_rx #252: Single SDR/Antenna, Multiple Sondes?

ZigiWalter, 2019-11-09:
Hi,

As discussed in other threads, there are cases in which a station is limited to a single SDR/antenna. In such setups, when multiple concurrent sondes can be received, it seems that AutoRX will only follow the strongest signal . Instead, one can consider a configuration option that will have AutoRX allocate the decoder on some course-grain alternating manner, allowing weaker sondes to be decoded too.

Technically, the existing code is so well organized that it seems very easy to modify it to achieve that (see proposal below). However, the real question is whether the authors think such enhancement belongs to the mainline AutoRX repository, or its considered a "niche use-case" that should better be a kept as a local modification (I remember reading that the intended use is multiple SDRs). 

Or maybe there is already a way to achieve that behavior in the existing implementation with no changes?

Thanks.

-----------------------------
The proposed change:
1. decode.py: The decoder quits after M minutes of decoding even if the sonde is still decoded successfully;
2. auto_rx.py: When a decoding task ends, its frequency is added to a (new) ordered list holding the "K-recently-decoded-frequencies";
3. scan.py: immediately after the peak_frequencies list is prepared (sorted by power, blocked frequencies removed, etc.) and is about to be tested for detection, any frequency that also appears in the aforementioned "K-recently-decoded-frequencies" list is moved to the end of the peak_frequencies list (not removed from it!).

This simply "penalizes" recently decoded sondes in the detection attempt order, which should result in round-robin allocation when multiple sondes are in-flight. Obviously, I might be missing things here...
--------------------------

darksidelemm, 2019-11-10:
This is similar to the 'skip mode' issue here: https://github.com/projecthorus/radiosonde_auto_rx/issues/72

I wouldn't necessarily implement it the way you suggest, and I would have it move through the sonde frequencies at a much faster rate, perhaps every 30 seconds. 

It would need to end up being a separate 'main loop' to auto_rx - implementing it in the current framework would be very challenging, and would likely lead to things being quite messy.

ZigiWalter, 2019-11-11:
Indeed, I can see obvious benefits in doing a single scan then multiple decodes (as opposed to multiple scans, multiple decodes in the proposed implementation). However, that would be much more difficult (at least, for me) to implement, as you indicated above.

So, I'll meanwhile keep this change in a side-repository (initial testing seems to have achieved the modest goals I was after), and hope that some day the modified "main loop" you have described above will make it into the mainline auto_rx.

Thanks!

joni73, 2020-01-15:
hi,

Please look at autowx2. I'am currently running autowx2 that start's radiosonde_auto_rx.
Good idea could be integrate radisonde_rx and autowx projects.

My problem fall's same gategory like yours but above is general solution to these.

I like to run at one or multiple SDR collecting satellites, takeing part to WSPR campaing to broader our understanding radio wave prediction as well support Radio Sonde tracking and have
weather satelites images. Also via DUMP1090 and from ARCAS messages it's possible
to get weather prediction related information.

Also provide accurate time and frequency standards and APRS support.

===
General solution need's complex prioritizing running proseses.
Physiaclly we might have SDR tongles's and antenna's that are difrent
 support to band, direction and sensitivyty as well accurancy (TXCO).

DIRECTION : sector, whit rotator, all around
ANTENNA BAND: 1G, 2M, 70cm,...
XTAL: no, TXCO, GPS syncronized PLL
SENSITIVITY: LNA, FILTERS, 

Then needed what we wanna get out as priority list.
This is hard part!

All that should be proccessed whit satellite passing,
baloon launch times to create time shedule to run
multiple dongles.

====

Structurally could be smart run and divide project:
-sheduller server that collects and creates shedule to RX process
-RX server that mainly saves data BASED ON monitoring bands
about trafic and select what signals will be recorded
-Analyze prosess that decodes data from RX system
-publishing module

This is quite near sturctures found rt-sdr, radiosonde_rx, autowx2, sattelite projects,...
autowx2 is trying to do this!

Problem is that we found same structures from ARCAS, radiosonde mosdules like
scanning frequencyes,... autowx is only time depend process starter not careing
about can trafic been heard!

TheSkorm, 2020-01-15:
@joni73 Autowx2 looks like a cool project though using autowx2 in radiosonde_auto_rx would likely over complicate the radiosonde_auto_rx project goals. I'd suggest that autowx2 use the lower level binaries provided by @rs1729 such as dft_detect for detection. Using radiosonde_auto_rx as a python module could possibly work, however we don't guarantee function compatibility between versions.

joni73, 2020-01-15:
It can be that eighter have to develop autowx2 to subsitidate radiosonde_auto_rx as you suggested
or do fork. Problem is that much code has overlaping functions after and before decoding. For example band scanning, prioritizing and reperetation and shareing data. Also that work is too big as small DIY fix. Some kind of DX_listener_scheduler_reporting allmost same as autowx2 or autowx2 fix needed.  At radiosonde / autowx2 integration I have not yet way to program "if radiosonde found do follow it". i can just do regular time's to listen. For ADB-S, ARCAS, radiosonde, satellites and wspr could be beneficial to share frequency scanning. Sanning and decoding frequency operations both have complex priorityes.  I'am still seeking keep it simple stubid approach solution. Anyone has smart guess what might work?

closed 2021-03-19 by darksidelemm