projecthorus/radiosonde_auto_rx #256: RS92 temperature

oe3osb, 2019-12-01:
Dears,
**RS92**: The temperature is shifted by one decimal place and the air pressure and humidity values are missing.

[image]

vy73 de Erich

darksidelemm, 2019-12-01:
... this is because the RS92 decoder used in auto_rx doesn't decode temperature or humidity data. 
-273.0 degrees C is used as a 'dummy' value.
If you can provide information on how temperature and humidity values can be decoded from the RS92 telemetry, I'm sure @rs1729 would be interested.

rs1729, 2019-12-01:
No, I have a version decoding RS92-PTU based on
https://brmlab.cz/project/weathersonde/telemetry_decoding
but I didn't include it in the github repository.
I think the website was updated since, have to take another look.
You say it would be nice to have it in the rs92-decoder?

darksidelemm, 2019-12-01:
I wouldn't say it's critical. RS92s are well and truly obsolete now, and I think by this point any stations still launching them are just running down old stock. Hopefully we won't see them for much longer.

rs1729, 2019-12-01:
I agree.
To include PTU from the old version in the new decoder would mean some work.
There is an Ozone-Station in Hohenpeissenberg/Germany that uses RS92, and the US RS92-NGP also have new serial numbers.

bazjo, 2019-12-01:
Hohenpeissenberg is running on a limited supply of all the sondes which remained throughout germany after the switch to rs41. NGP usage in the us should end in the next view years due to their move to become completly 400 MHz based

closed 2020-01-07 by darksidelemm

rs1729, 2021-02-05:
Added basic (uncorrected) RS92 PTU as described here:
https://github.com/rs1729/RS/issues/32