projecthorus/radiosonde_auto_rx #264: Speed <vel_h>, how to see?

sp3rat, 2020-02-23:
I am sorry my English.
What can I doing to see "Speed <vel_h>" sonde?
Is it possible?
Jurek de sp3rat

dterweij, 2020-02-23:
Using the UDP script you can see all values. 

horus_udp_receiver_example.py in auto_rx/utils map. Just add the payloads what you need to see.

Hint:  packet['speed']

sp3rat, 2020-02-24:
Many thanks  :)JurekWysłano z telefonu Samsung
-------- Oryginalna wiadomość --------Od: Danny Terweij <notifications@github.com> Data: 23.02.2020  23:48  (GMT+01:00) Do: projecthorus/radiosonde_auto_rx <radiosonde_auto_rx@noreply.github.com> DW: sp3rat <sp3rat@wp.pl>, Author <author@noreply.github.com> Temat: Re: [projecthorus/radiosonde_auto_rx] Speed <vel_h>, how to see?
  (#264) Using the UDP script you can see all values.
horus_udp_receiver_example.py in auto_rx/utils map. Just add the payloads what you need to see.
Hint:  packet['speed']

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFT6KAKHBWRGLFSWADLREL4J5A5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMWJVLQ#issuecomment-590125742",
"url": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFT6KAKHBWRGLFSWADLREL4J5A5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMWJVLQ#issuecomment-590125742",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

darksidelemm, 2020-02-24:
Still not quite sure i'm clear here - what's the issue? Where are you trying to see horizontal velocity data?

sp3rat, 2020-02-24:
Yes, I trying to see horizontal velocity data.Wysłano z telefonu Samsung
-------- Oryginalna wiadomość --------Od: Mark Jessop <notifications@github.com> Data: 24.02.2020  11:14  (GMT+01:00) Do: projecthorus/radiosonde_auto_rx <radiosonde_auto_rx@noreply.github.com> DW: sp3rat <sp3rat@wp.pl>, Author <author@noreply.github.com> Temat: Re: [projecthorus/radiosonde_auto_rx] Speed <vel_h>, how to see? (#264) Still not quite sure i'm clear here - what's the issue? Where are you trying to see horizontal velocity data?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFSJ5PP2P2J4JQRQ3ILREOM2DA5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMXH57Y#issuecomment-590249727",
"url": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFSJ5PP2P2J4JQRQ3ILREOM2DA5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMXH57Y#issuecomment-590249727",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

sp3rat, 2020-02-24:
I am trying ...Wysłano z telefonu Samsung
-------- Oryginalna wiadomość --------Od: Mark Jessop <notifications@github.com> Data: 24.02.2020  11:14  (GMT+01:00) Do: projecthorus/radiosonde_auto_rx <radiosonde_auto_rx@noreply.github.com> DW: sp3rat <sp3rat@wp.pl>, Author <author@noreply.github.com> Temat: Re: [projecthorus/radiosonde_auto_rx] Speed <vel_h>, how to see? (#264) Still not quite sure i'm clear here - what's the issue? Where are you trying to see horizontal velocity data?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFSJ5PP2P2J4JQRQ3ILREOM2DA5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMXH57Y#issuecomment-590249727",
"url": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFSJ5PP2P2J4JQRQ3ILREOM2DA5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMXH57Y#issuecomment-590249727",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

darksidelemm, 2020-02-24:
Ok, yes I see that this isn't on the web interface. It's going to take quite a bit of re-work to be able to show that on the web interface, as the table is already too wide as it is.

It's something I do plan on doing, it's just I haven't had much time to work on it...

If you set up Chasemapper it will show up there, so that might be an interim solution: https://github.com/projecthorus/chasemapper

sp3rat, 2020-02-24:
Ok Mark  :)Thank You very much.JurekWysłano z telefonu Samsung
-------- Oryginalna wiadomość --------Od: Mark Jessop <notifications@github.com> Data: 24.02.2020  11:35  (GMT+01:00) Do: projecthorus/radiosonde_auto_rx <radiosonde_auto_rx@noreply.github.com> DW: sp3rat <sp3rat@wp.pl>, Author <author@noreply.github.com> Temat: Re: [projecthorus/radiosonde_auto_rx] Speed <vel_h>, how to see? (#264) Ok, yes I see that this isn't on the web interface. It's going to take quite a bit of re-work to be able to show that on the web interface, as the table is already too wide as it is.
It's something I do plan on doing, it's just I haven't had much time to work on it...
If you set up Chasemapper it will show up there, so that might be an interim solution: https://github.com/projecthorus/chasemapper

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFUMTEAKBMCYUP3HPVTREOPHBA5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMXJ4JY#issuecomment-590257703",
"url": "https://github.com/projecthorus/radiosonde_auto_rx/issues/264?email_source=notifications\u0026email_token=ACXSXFUMTEAKBMCYUP3HPVTREOPHBA5CNFSM4KZ5GHC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMXJ4JY#issuecomment-590257703",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

samsonpwnz, 2020-03-07:
I didn't want to open a new "issue" for this. But it would be awesome to be able to get ascent/descent rate as well as direction and speed dumped to the log file that auto_rx creates.

darksidelemm, 2020-03-08:
I've been trying to avoid adding more and more guff to the log files.

There is a script which will process the log file and produce a sounding plot, if that's what you're after: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/plot_sonde_log.py

samsonpwnz, 2020-03-08:
When trying to run plot_sonde_log.py on a single file I get this, 
Read 5866 data points from 20200307-235001_R4530945_RS41_404300_sonde.log.
plot_sonde_log.py:239: RuntimeWarning: invalid value encountered in log
  DP = 243.04*(np.log(RH/100)+((17.625*T)/(243.04+T)))/(17.625-np.log(RH/100)-((17.625*T)/(243.04+T)))
plot_sonde_log.py:239: RuntimeWarning: divide by zero encountered in log
  DP = 243.04*(np.log(RH/100)+((17.625*T)/(243.04+T)))/(17.625-np.log(RH/100)-((17.625*T)/(243.04+T)))
plot_sonde_log.py:239: RuntimeWarning: invalid value encountered in double_scalars
  DP = 243.04*(np.log(RH/100)+((17.625*T)/(243.04+T)))/(17.625-np.log(RH/100)-((17.625*T)/(243.04+T)))
Detected burst at 35671 metres.

> 
> 
> I've been trying to avoid adding more and more guff to the log files.
> 
> There is a script which will process the log file and produce a sounding plot, if that's what you're after: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/plot_sonde_log.py

darksidelemm, 2020-03-08:
That's just suggesting its dealing with some bad values of RH (like, a RH of -1, which happens if the calibration data hasn't been received from the sonde yet). A similar error will be seen if the RH drops to 0 %.
It should just skip over these values and continue on.

samsonpwnz, 2020-03-08:
This is what I get for output, not sure if it is correct. 
http://puu.sh/FilBW/dc5061ccec.png

The reason I am after speed ,direction and vertical rate is our area just recently switched from the LMS-6 to the RS41. The RS41 seems to drop like a rock soon after burst, where as the LMS6 would stabilize to a nice slow descent. All this means the sondes just don't make it as close to me with out some serious jetstream winds to give them that push.

darksidelemm, 2020-03-08:
That plot looks like you're missing data for the first part of the ascent. That script expects data from near launch, so it probably won't be that useful to you...

It's sounding like they might be launching without parachutes. In that case a RS41 generally lands at about 9-10m/s descent rate. 

I should be able to add in some of the data you need into the log file. Just need to work out the best way to do it...

darksidelemm, 2020-03-20:
Have added vertical and horizontal velocity, and payload heading (which will be 180 degrees offset from standard wind direction) to the log file. This is currently in the testing branch, which will get merged into master once I've tested it out on my main station for a few days.

closed 2020-03-28 by darksidelemm

sp3rat, 2020-03-28:
Very, very gooood  :)
Thanks.

darksidelemm, 2020-03-28:
Just note that parsing the log file is a little bit broken at the moment due to the mismatch between the header line and some of the fields.

I'll fix that up sometime this week: https://github.com/projecthorus/radiosonde_auto_rx/issues/271