projecthorus/radiosonde_auto_rx #271: Add proper headers to logging output

darksidelemm, 2020-03-28:
The current log output tacks on 'optional' data at the end of each line, which breaks some CSV importers that expect the header to have the same number of fields as the data.

Suggest making a list of all the fields we want to see in the log file, and writing out dummy data (e.g. -1) if there is no data for that field available.

snh, 2020-04-15:
> Suggest making a list of all the fields we want to see in the log file, and writing out dummy data (e.g. -1) if there is no data for that field available.

Including all fields, including the optional ones, in the log file sounds like a good approach, and will ensure the structure is consistent 👍 

Writing nothing between the delimiters where a value doesn't exist is normally sufficient, ie

```
header1,header2,header3,header4,header5
field1,field2,field3,,field5
field1,field2,field3,field4,
```

darksidelemm, 2020-04-15:
Sounding good!
The extra fields start here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/logger.py#L138

snr, f_error, sats, batt, bt, aux

darksidelemm, 2020-07-24:
Resolved in https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.3.2

closed 2020-07-24 by darksidelemm