projecthorus/radiosonde_auto_rx #284: USB reset limit and error email notification, multiplexing

gmke, 2020-07-01:
auto_rx is a great project and I successfully run the sw since a few weeks (GALILEI, GALILEI_KL).  Thank you very much!
With one minor and one major improvement it would be an even better tool for me.

Unfortunately, I have frequent (ususally every second day) USB problems with my two receiver setup. auto_rx does the clever trick of trying to reset the USB device, but this does not help sometimes. Not so clever, it tries to repeat this trick until eternity. It would be good to stop this after a few tries and send an error notification by email (like it does for detections).

We have a lot of radiosondes here in Central Europe, if auto_rx would support a multiplexing schedule, a lot more sondes could be detected and followed, although with reduced time resolution. With two receivers, one could always scan and the other(s) would switch between the list of available sondes every few seconds. Even with a single receiver, there are options. With three RTL-SDR receivers, three 2Mhz bandwidth chunks could be sampled and decoded for every sonde receivable.

darksidelemm, 2020-07-02:
The USB reset limit is probably worthwhile, but it may end up just resulting in a station which stops working after some time. I can't really do much about this, it's just some bug in the RTLSDR and/or driver. 

As for the 'multiplexing' mode, it's already mentioned in other issues, like this one: https://github.com/projecthorus/radiosonde_auto_rx/issues/72

I understand why it's wanted, but it does mean a lot of re-work in auto-rx, and a lot of data missed due to the time required to start/stop the receiver/demodulator processes. It's not something I've really had the time or inclination to work on, preferring to improve demodulator performance and add support for new sondes. 
TBH I would prefer a solution that results in all scanned sondes being demodulated simultaneously, but this means a lot of work to produce a multi-client SDR server system, similar to spyserver, but not closed source.

gmke, 2020-07-02:
the station stops working already because right now it does not switch to using the remaining good stick but tries to reset the faulty one forever. Therefore I believe it would be very worthwhile to change this behaviour and let the operator know by sending an email. 
Alternatively you could start using syslogd and write the detection messages to INFO and the USB resets to ERROR. Then the admin can configure syslogd to send emails as he wishes. Btw this design would be more inline with the UNIX way of doing things.

ZigiWalter, 2020-07-03:
Regarding time multiplexing raised by the OP - as a poor man's approximation, I can propose this: https://github.com/projecthorus/radiosonde_auto_rx/issues/252#issue-520479217

Obviously, this is inferior to what @darksidelemm described aobve, but maybe addresses a part of the need.

rs1729, 2020-07-03:
https://github.com/projecthorus/radiosonde_auto_rx/issues/152
I hear that it is possible to use
https://github.com/rs1729/RS/tree/test/demod/multi/
in auto_rx udp-mode. Though udp-mode is probably for people who know what they are doing.
In rs_mutli you can add and remove signals via a fifo-file, i.e. the number of signals is not fixed. But for a new scan it would have to be stopped or another rtl-sdr (with known frequency offset difference) for scanning would be needed.
At least for frequency-stable RS41 and standard launch times this could be an option.
For autonomous operations with a single rtl-sdr the update-scans would have to use the same data as the demodulators that prepare the data for the decoders. You could stop decoding for scanning, but for e.g. for RS41 it would  recommendable to save the calibration data.
A quick solution could be a monitoring (or signal hopping) option such that the next signal in the scan results is decoded after every 2 minutes or so.

darksidelemm, 2020-07-03:
I've started progressing this a bit in the testing branch.

SDRs will now be removed from the SDR list after they fail to recover after 5 reset attempts. Eventually, if there are no SDRs available, auto_rx will exit completely.

I still need to clean up the logging outputs a bit, but it's probably possible to use systemd to send an email when a service exits / restarts. 

I need to rework the Email Notification module a bit to make it easier to send arbitrary emails, then I'll add an e-mail notification as well (assuming the user has supplied suitable details).

darksidelemm, 2020-07-24:
This has been partially resolved in https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.3.2

SDRs will drop off the list as they fail, but there's no e-mail notification yet.

darksidelemm, 2020-12-28:
https://github.com/projecthorus/radiosonde_auto_rx/wiki/RS----Auto_RX-JSON-Formats

closed 2020-12-28 by darksidelemm