projecthorus/radiosonde_auto_rx #286: Multiple SDRs - Start in config order

Heppie, 2020-07-03:
I have 2 SDRs, 1 with a bias-t support and 1 without, connected and configured in station.cfg as follows:

[sdr_1]
device_idx = 00000102
bias = true
[sdr_2]
device_idx = 00000101
bias = false

When starting auto_rx it seems to use them in Serial number order rather than config order which prevents the bias-t enabling.

Yes I know I could change the Serial numbers to resolve this but it would be nice to be able to define the order in the station.cfg expecially if you have >2 SDRs

darksidelemm, 2020-07-03:
I'm not sure what the problem here is?  Is one of the SDRs powering a preamp which is split between both SDRs?

auto_rx uses the SDRs as a 'pool'. Any SDR can be used for any task, be it scanning or decoding a sonde.

Heppie, 2020-07-03:
You are correct. I have an LNA infront of a power splitter which is powered by the bias-t on one of my SDRs.

My issue is that the bias-t does not turn on until auto_rx has used the SDR for scanning. Unfortunately even though it is sdr_1 in config, auto_rx picks the other SDR (I beleive it is doing it on serial number although it may be device index) which does not have a bias-t. Once it has been used for scanning the bias-t stays on until exit of auto_rx so is fine when it switches back and forth between scanning, decoding and idle.

darksidelemm, 2020-07-04:
This is, unfortunately, the expected behaviour, and SDR allocation does work off serial number initially, but will end up just picking whichever SDR is free.

It's not really possible to pre-determine all the different setups of SDRs and splitters that might be possible here, so trying to force certain orderings is not something I ever had in mind. 

I kind of expected that in a multiple-SDR-one-antenna case, any LNA would be powered by a *common* bias tee, most likely on the antenna side of the RF splitter (since most RF splitters do not pass DC - yes I know there are ones with DC pass through).

So in your case, either switch the SDRs around (probably the simplest fix), or use a separate bias tee.

darksidelemm, 2020-07-04:
Actually looking more closely at the code, the allocate_sdr function is here:
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/auto_rx.py#L74

It looks like I'm just iterating through sdr_list.keys(), which doesn't necessarily have a guarantee of being in order. I can probably make that at least go through in an ordered sense (which will be based on the serial), but thats probably about the best I can do here.

darksidelemm, 2020-07-24:
Resolved in https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.3.2

closed 2020-07-24 by darksidelemm